OctoPrint/OctoPrint #4480: [RC Feedback] Feedback on the 1.8.0rc4 Release Candidate

foosel, 2022-04-05:
# 1.8.0rc4 isn't release yet, this is a placeholder ticket

Please provide *general* feedback on your experience with the 1.8.0rc4 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

pingywon, 2022-04-05:
Upgraded from RC3 and everything appears to be working as expected.

goeland86, 2022-04-05:
Trying to update from 1.7.3 to 1.8.0-RC4 (had to reflash my initial image, hence why I'm not on RC3 anymore), I get the following error in octoprint.log:
```
2022-04-05 15:37:04,857 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 2090, in _perform_update
    else:
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 115, in perform_update
    _log_message(
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/pip.py", line 208, in execute
    CommandlineCaller.__init__(self)
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/commandline.py", line 244, in call
    # thread
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/commandline.py", line 236, in process_stdout
    }
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/commandline.py", line 230, in process_lines
    kwargs.update(
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/pip.py", line 498, in _preprocess_lines
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/pip.py", line 517, in _preprocess
    # This is horribly ugly and I'm sorry...
  File "/home/debian/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return to_bytes(_ANSI_REGEX.sub("", to_unicode(line)))
UnicodeEncodeError: 'latin-1' codec can't encode characters in position 5-44: ordinal not in range(256)
2022-04-05 15:37:04,918 - octoprint.plugins.softwareupdate - INFO - Saved update log to disk
2022-04-05 15:37:05,071 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'target': 'octoprint', 'from_version': '1.7.3', 'to_version': '1.8.0rc4'}
2022-04-05 15:38:19,397 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-04-05 15:38:42,616 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 109824, 'printer_state': 'OPERATIONAL'}
```
In the plugin_softwareupdate_console.log I get this content only:
```
2022-04-05 15:35:54,382   /home/debian/OctoPrint/venv/bin/python -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint/archive/1.8.0rc4.zip --no-cache-dir
2022-04-05 15:35:59,448 > Collecting https://github.com/OctoPrint/OctoPrint/archive/1.8.0rc4.zip
2022-04-05 15:36:00,880 > Downloading https://github.com/OctoPrint/OctoPrint/archive/1.8.0rc4.zip
2022-04-05 15:36:00,882 > | 6.6 MB 7.2 MB/s 0:00:00
2022-04-05 15:36:02,888 > Preparing metadata (setup.py): started
```

goeland86, 2022-04-05:
> Trying to update from 1.7.3 to 1.8.0-RC4 (had to reflash my initial image, hence why I'm not on RC3 anymore), I get the following error in octoprint.log:
>
> ```
> 2022-04-05 15:37:04,857 - [...]

Oddly enough, different upgrade attempt on a separate pi, and the upgrade went through. Restarted the system and tried again. Update went through as well. Possibly a corrupt zip download? Sorry for spamming the thread.

b-morgan, 2022-04-05:
Upgraded without issue.

foosel, 2022-04-05:
@goeland86 known issue, see https://github.com/OctoPrint/OctoPrint/issues/4392. Incompatibility with latest pip, downgrade or just restart as it probably succeeded in the background.

1n5aN1aC, 2022-04-06:
When I upgraded to rc4, the webui broke visually for me.
Basically, font awesome was failing to load.  web console was showing it failing to fetch the resources.  (Reloading without cache did not help)

I rebooted the system, (octopi on a ras4) and reloaded the page and all was good.

Other than that, everything seems to be working perfectly fine!

foosel, 2022-04-06:
Please share a system info bundle so this can be investigated.

1n5aN1aC, 2022-04-06:
> Please share a system info bundle so this can be investigated.

Gladly.  See here for bundle download:  https://wormhole.app/4WydX#2hgRsXXiErcL7l9Xz30v3w

foosel, 2022-04-06:
Resharing here so we can load it into the bundle viewer more easily (please upload into the comment in the future instead of using a third party file share service)

[octoprint-systeminfo-20220406185639.zip](https://github.com/OctoPrint/OctoPrint/files/8429787/octoprint-systeminfo-20220406185639.zip)

foosel, 2022-04-06:
Sadly there was a log rollover after you updated, so the log from the error isnt there :/ Can you please provide the log from before the rollover if it is still there? You can download it via Settings > Logging, it should be named something like octoprint.log.1 if I remember correctly.

1n5aN1aC, 2022-04-06:
I didn't find it there, but I did find it at /home/pi/.octoprint/logs

[octoprint.log.2022-04-05.txt](https://github.com/OctoPrint/OctoPrint/files/8430061/octoprint.log.2022-04-05.txt)

ChrisHeerschap, 2022-04-06:
Been upgrading the RCs as the notifications come up, and been printing regularly on a variety of raspi systems (3 and 4) connected to a couple Prusas, MK3S and Mini, no problems at all.

foosel, 2022-04-07:
@1n5aN1aC Sadly nothing in there that suggests anything was amiss, just a normal update. The reboot you mentioned also doesn't show in there though.

1n5aN1aC, 2022-04-07:
> The reboot you mentioned also doesn't show in there though.

Yeah sorry, I muddied the waters by experiencing that on the 5th, shrugging and moving on with life, then coming back to it on the 6th, rebooting it (solving the problem), and then printing several models.

fcchambers, 2022-04-08:
Update from Stable Release to RC4:  **Executed flawlessly**
Printing known good GCode: Awful, stuttering, frequent pauses, over extrusions on every layer
**Disabled all plug-ins, disconnected web cam: Printed flawlessly.**

So *something* has an issue with RC4, but it doesn't look to be an OctoPrint issue at all. :)

oliof, 2022-04-08:
RC4 works fine for me on an Ultimaker Original running a current Marlin build.

foosel, 2022-04-11:
Looks like I'll somehow have to carve out the time needed to push out another RC this week because there's still a bad bug in the settings refactoring that I just fixed. Wheee.

foosel, 2022-04-12:
1.8.0rc5 is out, new feedback ticket is #4487.

closed 2022-04-12 by foosel