OctoPrint/OctoPrint #4481: [Request] allow to select part which should not be printed while printing

rdmueller, 2022-04-05:
### Is your feature request related to a problem? Please describe.

Do you also have some time the problem that you print multiple parts at once and after some hours you notice that one of those parts is broken and will break you whole print? 

### Describe the solution you'd like

I think it could help to be able to mark a region in the G-Code-Viewer to be skipped. You should be able to do this while the print is already running in order to try to safe the rest of your print from getting jammed.

### Describe alternatives you've considered

yes, I could print all parts by themselves or I could try to slice my parts in such away that each part is printed after the other. Both solutions have their own problems.

foosel, 2022-04-05:
There's already a plugin for that ;)

https://plugins.octoprint.org/plugins/cancelobject/

foosel, 2022-04-05:
Make that two actually: https://plugins.octoprint.org/plugins/excluderegion/

closed 2022-04-05 by foosel

rdmueller, 2022-04-05:
wow. that was quick. Thanx for your Help!