OctoPrint/OctoPrint #4482: [Request] Support For adding a redundant sensor reading to the tempurature chart

StreckerCM, 2022-04-05:
### Is your feature request related to a problem? Please describe.

I am running a redundant temp sensor on my system.  Marlin has support for auto reporting a redundant temp sensor reading but it is disabled by default.

### Describe the solution you'd like

When enabled, the redundant temperature sensor is passed in the auto report string in this format 'R: XXX.XX / XXX.XX'.  The first set of temperature values is the redundant sensor reading.  The second set is the reading of the primary sensor.  In this example, Bed (B) is the sensor that is configured as redundant.

>  T:24.00 /0.00 B:41.41 /0.00 C:21.09 /0.00 R:38.27 /41.41 @:0 B@:0
>  T:23.86 /0.00 B:41.71 /0.00 C:21.08 /0.00 R:38.15 /41.71 @:0 B@:0
>  T:24.22 /0.00 B:41.57 /0.00 C:20.95 /0.00 R:38.13 /41.57 @:0 B@:0
>  T:24.00 /0.00 B:41.52 /0.00 C:20.84 /0.00 R:38.08 /41.52 @:0 B@:0
>  T:24.16 /0.00 B:41.82 /0.00 C:21.18 /0.00 R:38.21 /41.82 @:0 B@:0
>  T:23.83 /0.00 B:41.38 /0.00 C:21.16 /0.00 R:38.17 /41.38 @:0 B@:0
>  T:24.02 /0.00 B:41.38 /0.00 C:21.06 /0.00 R:38.16 /41.38 @:0 B@:0
>  T:23.91 /0.00 B:41.43 /0.00 C:20.98 /0.00 R:38.09 /41.43 @:0 B@:0

I would like to see an option added to the printer profile to enable the parsing of a redundant sensor.  There should also be a box to enter a custom label for the redundant sensor.

[image]

When enabled, Octoprint should look to see if the temperature string contains the R-value and if it does, it should plot it. 

### Describe alternatives you've considered

I tried to find a plugin that would support this but there was nothing in the repository.

cp2004, 2022-04-05:
In OctoPrint 1.8.0 (currently available in RC) there is support for parsing the extra temperature values, for use in plugins etc. Combined with the [Plotly Temperature Graph](https://plugins.octoprint.org/plugins/plotlytempgraph/) this can be shown on the graph.

jneilliii, 2022-04-05:
I think even the default temp graph bundled with 1.8.0 added support for the additional reported data.

StreckerCM, 2022-04-07:
> In OctoPrint 1.8.0 (currently available in RC) there is support for parsing the extra temperature values, for use in plugins etc. Combined with the [Plotly Temperature Graph](https://plugins.octoprint.org/plugins/plotlytempgraph/) this can be shown on the graph.

Thanks, I installed the RC and confirmed that this is the case.  I'll close this item

closed 2022-04-07 by StreckerCM