OctoPrint/OctoPrint #4487: [RC Feedback] Feedback on the 1.8.0rc5 Release Candidate

foosel, 2022-04-12:
# 1.8.0rc5 isn't released yet, this is a placeholder ticket

Please provide *general* feedback on your experience with the 1.8.0rc5 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

b-morgan, 2022-04-12:
Installed (from RC4) with no issues.

pingywon, 2022-04-12:
Updated from RC4 with out issue. Anything is particular you would like tested @foosel ?

1n5aN1aC, 2022-04-12:
Updated from RC4 just fine, clicked around, changed some minor settings, started print, all seemed fine.

Also, the icons disappearing didn't happen this time, so we're good there.

fcchambers, 2022-04-13:
Updated from most recent stable release, no issues. Printing fine with no third party plugins.

As with RC4, I've encountered problems (Stuttering printing, frequent pauses, globs on pauses) with a plug in, I have it narrowed down to one of two plugins, will figure out which one and open an issue with them.

ckuethe, 2022-04-14:
All is working for me so far, upgraded from release.

rPi 3b+, Ender3, UVC webcam, sent gcode via API from PrusaSlicer
Plugins: ArcWelder, BLTouch, Bed Visualizer, MeatPack, Printer Notifications, Marlin GCode Documentation, Camera Settings, Marlin EEPROM Editor, Preheat Button, PrettyGCode, ...

Guilouz, 2022-04-17:
Some strings are not translated when we use language file (all categories names on the left in settings).

[image]

cp2004, 2022-04-17:
That is down to the individual language pack you are using (probably) not adding those as translated strings. The only one that bundles with OctoPrint is German, and that seems to be working fine.

Guilouz, 2022-04-17:
@cp2004 my french language file is up to date (all strings translated at 100%) based on this .pot file : https://github.com/OctoPrint/OctoPrint/blob/rc/maintenance/translations/messages.pot

[image]

fcchambers, 2022-04-22:
Starting week two of using RC5 as my "daily driver" on an Ender 3Pro. GREAT results so far.

b-morgan, 2022-05-09:
I opened OctoPrint this morning and it tells me that an update to 1.8.0rc4 is available. I running 1.8.0rc5 so I don't know what this means?

cp2004, 2022-05-09:
@b-morgan the best theory we can think of is some GitHub issue, the tag still existed but wasn't showing up in latest releases. It should be working now, but the software update will have a 12 hour cache if you don't force-check for an update.

foosel, 2022-05-10:
Can definitely confirm that I did not unpublish the RC or anything like that. I didn't even touch the release page since pushing RC5. I guess there indeed was some weird hiccup, I sure hope though that this won't repeat itself.

cp2004, 2022-05-10:
GitHub confirmed it was their issue and that they didn't believe it would happen again in the ticket I opened yesterday.

b-morgan, 2022-05-10:
Can we follow this issue on GitHub (i.e. what is the ticket number)?

cp2004, 2022-05-10:
> Can we follow this issue on GitHub (i.e. what is the ticket number)?

The only info I have is from my private issue report ticket, I am not aware of a public ticket existing at the moment.

closed 2022-06-16 by cp2004