OctoPrint/OctoPrint #4488: Compatibility with Python 3.11

cp2004, 2022-04-12:
### Is your feature request related to a problem? Please describe.

Python 3.11 is currently in alpha, scheduled to be released towards the end of the year. I tested it to see if OctoPrint runs successfully.

There were not as many issues to solve as there was with 3.10, which is good.

### Describe the solution you'd like

The only change I've found so far that needs to happen is for wrapt to be bumped to a newer release (fixing some change to a deprecated API) - 1.14.0.

However... according to the issues over there, the current version of wrapt doesn't work on 3.7 (and maybe others?). It looks like the issues are under control (https://github.com/GrahamDumpleton/wrapt/issues/210) so we will need to wait for a newer release, hopefully soon.

So I created this issue so it was not forgotten 🙂

### Describe alternatives you've considered

🤷 

### Additional context

https://github.com/GrahamDumpleton/wrapt/issues/210

https://github.com/GrahamDumpleton/wrapt/issues/196

```
(venv311) PS C:\Users\charl\Development\OctoPrint Core\OctoPrint> pip install -e .
Obtaining file:///C:/Users/charl/Development/OctoPrint%20Core/OctoPrint
  Preparing metadata (setup.py) ... done
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in c:\users\charl\development\venv311\lib\site-packages (from OctoPrint==2.0.0.dev1755+gf0ed6ede6) (2021.2.23)

[ removed ]

Collecting wrapt<1.14,>=1.13.3
  Using cached wrapt-1.13.3.tar.gz (48 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [92 lines of output]
      Traceback (most recent call last):
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 35, in __getattr__
          return next(
                 ^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 36, in <genexpr>
          ast.literal_eval(statement.value)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python311\Lib\ast.py", line 108, in literal_eval
          return _convert(node_or_string)
                 ^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python311\Lib\ast.py", line 107, in _convert
          return _convert_signed_num(node)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python311\Lib\ast.py", line 81, in _convert_signed_num
          return _convert_num(node)
                 ^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python311\Lib\ast.py", line 72, in _convert_num
          _raise_malformed_node(node)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python311\Lib\ast.py", line 69, in _raise_malformed_node
          raise ValueError(msg + f': {node!r}')
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      ValueError: malformed node or string on line 2: <ast.Call object at 0x000002A0FA470190>

      The above exception was the direct cause of the following exception:

      Traceback (most recent call last):
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 419, in _parse_attr
          return getattr(StaticModule(module_name), attr_name)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 43, in __getattr__
          raise AttributeError(
          ^^^^^^^^^^^^^^^^^^^^^
      AttributeError: wrapt has no attribute __version__

      During handling of the above exception, another exception occurred:

      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\charl\AppData\Local\Temp\pip-install-x1mo5zr4\wrapt_c71b76722d194d95ac334f241d74ba7f\setup.py", line 41, in <module>
          setuptools.setup(
          ^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\__init__.py", line 155, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\_distutils\core.py", line 122, in setup
          dist.parse_config_files()
          ^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\dist.py", line 818, in parse_config_files
          parse_configuration(
          ^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 158, in parse_configuration
          meta.parse()
          ^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 498, in parse
          section_parser_method(section_options)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 469, in parse_section
          self[name] = value
          ~~~~^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 222, in __setitem__
          value = parser(value)
                  ^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 599, in _parse_version
          version = self._parse_attr(value, self.package_dir)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python311\Lib\contextlib.py", line 155, in __exit__
          self.gen.throw(typ, value, traceback)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 55, in patch_path
          yield
          ^^^^^
        File "C:\Users\charl\Development\venv311\Lib\site-packages\setuptools\config.py", line 422, in _parse_attr
          module = importlib.import_module(module_name)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python311\Lib\importlib\__init__.py", line 126, in import_module
          return _bootstrap._gcd_import(name[level:], package, level)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
        File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
        File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
        File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
        File "<frozen importlib._bootstrap_external>", line 924, in exec_module
        File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
        File "C:\Users\charl\AppData\Local\Temp\pip-install-x1mo5zr4\wrapt_c71b76722d194d95ac334f241d74ba7f\src\wrapt\__init__.py", line 10, in <module>
          from .decorators import (adapter_factory, AdapterFactory, decorator,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\charl\AppData\Local\Temp\pip-install-x1mo5zr4\wrapt_c71b76722d194d95ac334f241d74ba7f\src\wrapt\decorators.py", line 34, in <module>
          from inspect import ismethod, isclass, formatargspec
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      ImportError: cannot import name 'formatargspec' from 'inspect' (C:\Program Files\Python311\Lib\inspect.py)
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

foosel, 2022-05-11:
Marked as task and tentatively scheduled for 1.9.0 (more to have it on recall than anything else really).

kevinsumner, 2022-10-31:
I've been working on bringing up a machine with Fedora 37 to test compatibility with the various software packages I use. F37 has upgraded to Python 3.11 and its release is scheduled for the next couple of weeks ([schedule](https://fedorapeople.org/groups/schedule/f-37/f-37-key-tasks.html)). OctoPrint 1.8.6 does not have #4530, so the pip install instructions fail on F37.

Would it be possible (and appropriate) to cut a 1.8.x release with #4530?

foosel, 2022-10-31:
Sorry, but no. OctoPrint 1.8.x clearly states it is compatible to 3.7 through 3.10. Releases cost time. Usually half a day to a day, plus a TON of support overhead for everyone helping out in the following days due to "I updated and now everything is broken, the update is shit" red herrings. I do them for fixes (specifically regressions and vulnerabilities), but a software not running under a runtime environment it doesn't explicitly say it supports is not a regression, not even a bug, it's working as designed. And every single 1.8.x release means yet another delay in getting around to working on 1.9.0. 

I'm pretty sure even Fedora supports creating a virtual environment with an older Python version than the new system wide default. If that isn't an option, you can also just run a checkout of `maintenance`.

kevinsumner, 2022-10-31:
You have no reason to be sorry. I recognized while writing my last comment that releases for every project come with work amount of work that someone must do, but how much varies depending on the project. Your time is valuable and I want to try to respect that. I can tell from your response that you've dealt with people demanding support (particularly free support) for a project they've never contributed to in any way.

I appreciate the guidance on venv and maintenance. I'll use one of those for now.

flaviut, 2022-11-22:
Note that this can also be worked around by manually installing the correct wrapt afterwards:

```
source venv/bin/activate
pip install 'wrapt>=1.14,<1.15'
```

Pip will complain about unmet dependencies, but it will work fine. This is the approach I took when I switch my Octoprint installation [to Python 3.11 under Debian Bullseye](https://github.com/flaviut/bullseye-python3.11-armhf).

sammcj, 2023-02-10:
While it's not pretty I can confirm that @flaviut's workaround of installing `'wrapt>=1.14,<1.15'` works on Armbian 11.6 with Python 3.11.2 - however - after Octoprint starts, it's configuration seems normal but no plugins are found so I wouldn't recommend relying on it.

Hopefully there's a proper fix for OctoPrint with Python 3.11 soon. 🤞

closed 2023-03-02 by foosel