OctoPrint/OctoPrint #4489: Webcam not Viewable on OctoPrint site when accessing via reverse Proxy, but viewable when accessing pi via IP Address

RFBomb, 2022-04-15:
### The problem

First off, this is NOT a major issue, more a slight inconvenience than anything.
My Setup:
- I use a home server with a docker instance that runs the Heimdall container, and uses a reverse proxy (LinuxServer/Swag) to point to the other contains and network things I want exposed (password protected obviously). 
- All of this is accessible to my laptop and phone using a DuckDNS domain.
- Octoprint is running off a raspberry pi connected to my printer.
- Heimdall points to my octoprint device using the following pattern: ` https://[MyDomain].duckdns.org/octoprint/`

When accessing Octoprint via this setup, everything works fine, except I cannot view the webcam. It reports this message:
 
[image]

When I am on my local network however, the webcam can be viewed if I access the octoprint server directly using the IP address.

Heres where it gets interesting I think:
- I use Cura to slice and connect to Octoprint via the DuckDNS link. Cura streams the webcam just fine. 
- I use the 'Polymer' app on my iphone via the DuckDNS link, which also streams the webcam just fine. 
- If I change the webcam url on octoprint from the default to `octoprint/webcam/?action=stream`, then I can now view it on the octoprint server via duckdns, but Cura, the app, and accessing it directly via its local IP address all lose the stream. 

Further, clicking on the link in the image shown above provides this link in the url: `https://[MyDomain].duckdns.org/webcam/?action=stream` which results in a 404.
Updating the link to: `https://[MyDomain].duckdns.org/octoprint/webcam/?action=stream` results in the stream.

I find this issue quite odd, as I would suspect the server display itself would autodetect the stream path based on the access point. If it does this already, it appears to only be evaluating the initial domain, and ignoring the 'subfolders' (i can't recall the actual term at the moment) of the domain. I say this, because If I recall correctly, when I had it set up for `Octoprint.[MyDomain].DuckDns.Org` the stream would work fine (because it was directly following the `.org` portion of the address)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

Version 0.18.0, running on Raspberry Pi 4 Model B Rev 1.1

### Printer model & used firmware incl. version

Ender 3

### Browser and version of browser, operating system running browser

Firefox / all

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220415110250.zip](https://github.com/OctoPrint/OctoPrint/files/8496618/octoprint-systeminfo-20220415110250.zip)
[image]

fieldOfView, 2022-04-15:
This is more suitable for the community forum here:
https://community.octoprint.org

OctoPi comes with a reverse proxy of its own (haproxy), which "redirects" the /webcam requests to port 8080, where mjpgstreamer lives. The reverse proxy you added redirects all traffic straight to OctoPrint (at local port 5000), thus bypassing the proxy that provides the /webcam redirect. If you have your reverse proxy forward traffic to port 80 instead of port 5000, you hand it over to haproxy which then correctly handles the /webcam and forwards the other requests to OctoPrint.

cp2004, 2022-04-15:
Your webcam address is set as `/webcam/?action=stream`, so this is appended to the domain name by the browser, not including `/octoprint`. When you are proxying it through `/octoprint` on the end you'll need to change the stream URL accordingly. You could use the full name (incl. https://domain.duckdns.org etc) so it will work anywhere, but you will miss local control.

Regardless, this is not an issue with OctoPrint so it is not for the OctoPrint issue tracker. For help and support in using OctoPrint, plugins, and your general setup please get in touch on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org) - helpful people are around that can answer your questions better than the handful of people monitoring here.

closed 2022-04-15 by cp2004

RFBomb, 2022-04-24:
> The reverse proxy you added redirects all traffic straight to OctoPrint (at local port 5000), thus bypassing the proxy that provides the /webcam redirect. If you have your reverse proxy forward traffic to port 80 instead of port 5000

My proxy detects the url `..DuckDNS.org\octoprint\` and redirects all traffic to 192.168.1.105:80, so I don't know understand why the octoprint proxy isn't working for this situation. 

Once again, all apps and external programs that connect through that address work fine for both control and webcam viewing, except for the octoprint-hosted web page itself. (Yet accessing that same page using the lan address works fine).

So I think it's an issue where it isn't properly setting up/detecting where it lists to for the camera.  
Maybe the `/octoprint` in the proxy url is confusing the octoprint proxy somehow (such as if /octoprint is already specified by the proxy? Not sure)

RFBomb, 2022-04-24:
> Your webcam address is set as /webcam/?action=stream, so this is appended to the domain name by the browser, not including /octoprint

This is what I'm getting at though. Why wouldn't it just take whatever the access url is and append it? Why extract the domain and append?

cp2004, 2022-04-24:
All the proxies are setup correctly, since you can access the webcam (just not in the OctoPrint page).

> > Your webcam address is set as /webcam/?action=stream, so this is appended to the domain name by the browser, not including /octoprint
> 
> This is what I'm getting at though. Why wouldn't it just take whatever the access url is and append it? Why extract the domain and append?

That's how the web browser works. OctoPrint is not the one in control here, it gives the URL you have configured straight to the browser in an `<img>` tag. It's not up to us what it does, it is the web standard.

When you give it a link to `/webcam`, it is starting with a `/`, or the 'root' of the website. When you host it at `/octoprint`, it is not at the root anymore, but your link is still to the root + webcam.

You could try just `webcam/?action=stream` without the leading slash. That should make the browser then just directly add it on, not at the website root.

If you have further questions please use the community forums or discord server as mentioned, since it's not an issue with OctoPrint.