OctoPrint/OctoPrint #4493: The first M20 done on connection is missing the L flag.

nophead, 2022-04-22:
### The problem

With the latest version of Marlin with `Cap:EXTENDED_M20:1` the first display of the file list after connection does not show long filenames but refreshing the list does. The first M20 command that is sent immediately after the `M115` does not have the `L` parameter but subsequent ones do. 

It looks like the M20 command is built and added to the send queue before the results from the M115 have been parsed and the EXTENDED_M20 capability set.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.7.3 Python 3.7.3

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Mendel90

### Browser and version of browser, operating system running browser

Chrome on Win7 64.

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220422153035.zip](https://github.com/OctoPrint/OctoPrint/files/8541814/octoprint-systeminfo-20220422153035.zip)

```
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:HOST_ACTION_COMMANDS:0

---------------------------------------------------------------------------
Your printer's firmware supports host action commands, but they are
disabled. The firmware will not inform OctoPrint about job pause &
cancellations, filament runouts, and similar firmware-side events. If you
want this functionality, you need to change your firmware's configuration.

Learn more at https://faq.octoprint.org/warning-firmware-hostcommands
---------------------------------------------------------------------------

Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:LFN_WRITE:1
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0
Recv: Cap:EXTENDED_M20:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: ok
Send: M20
Recv: Begin file list
Recv: TOPCAP_E.G 1099868
Recv: USB_CA~1.GCO 511242
Recv: LOWERB~1.G 3668226
Recv: End file list
Recv: ok
Send: M155 S2
Recv: ok
[...]
Send: M20 L
Recv: Begin file list
Recv: TOPCAP_E.G 1099868 TOPCAP_E.G
Recv: USB_CA~1.GCO 511242 USB_CA~1.GCO
Recv: LOWERB~1.G 3668226 lowerbrkt_export.g
Recv: End file list
Recv: ok
[...]
```

jneilliii, 2022-04-30:
Also while printing the shortened file name is used in state panel/job status name for file. Should that be a separate issue or should we roll these into the same issue?

[image]

cp2004, 2022-04-30:
I took a quick peek, looks like the frontend may be thinking it's using the display name but that is wrong when it comes out of the socket from server. Probably separate issues.

cp2004, 2022-06-19:
## Implementation help
*If a contributor wants to tackle this issue, some directions on where to start*

* When the capability report is received, then the SD card long file names could be queried, similar to how the autoreporting intervals are set. A comparison on how they are implemented and do things similarly would be good. https://github.com/OctoPrint/OctoPrint/blob/53b9b6185781c07e8c4744a6e28462e96448f249/src/octoprint/util/comm.py#L2699-L2723
* [`refreshSdFiles` function](https://github.com/OctoPrint/OctoPrint/blob/53b9b6185781c07e8c4744a6e28462e96448f249/src/octoprint/util/comm.py#L1992)

foosel, 2022-09-28:
PR merged (thanks @kForth!), this will be part of 1.9.0

closed 2023-05-23 by foosel

50n1c, 2023-06-02:
Octoprint 1.9.0 still sends M20 command instead of M20 L in Marlin 2.1.2 with EXTENDED_M20 support.
```
Recv: Marlin 2.1.2
Recv: echo: Last Updated: 2022-12-17 | Author: (Son1c, Ender 3)
Recv: echo: Compiled: May 23 2023
Recv: echo: Free Memory: 10505  PlannerBufferBytes: 1344
Recv: //action:notification Ender 3 Ready.
Recv: echo:V87 stored settings retrieved (630 bytes; crc 41669)
Recv: //action:notification Stored settings retrieved
Recv: //action:prompt_end
Recv: echo:SD card ok
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.1.2 (May 23 2023 18:56:16) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender 3 EXTRUDER_COUNT:1 
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:1
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:1
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:HOST_ACTION_COMMANDS:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:LFN_WRITE:1
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:1
Recv: Cap:EXTENDED_M20:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:0
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: ok
Send: M21
Recv: echo:SD card ok
Recv: ok
Send: M117 Octoprint connected.
Recv: //action:notification Octoprint connected.
Recv: ok
Send: M20
Recv: Begin file list
Recv: VOBLIN~1.GCO 706417
Recv: 413_VO~1.GCO 669700
Recv: 36_VOB~1.GCO 679729
Recv: PRUSA_~1.GCO 589006
Recv: 413_CH~1.GCO 465824
Recv: PRUSA_~2.GCO 541537
Recv: End file list
Recv: ok
Send: M155 S2
Recv: ok
Send: M876 P1
Recv: ok
Recv:  T:28.89 /0.00 B:28.78 /0.00 @:0 B@:0
```
In Octoprint settings, the option "Send M20 L instead of M20 to use extended SD file lists, if detected as supported by the firmware" is also enabled.

foosel, 2023-06-05:
Make sure you have "Wait to load the SD card file list until the firmware capability report is processed" enabled:

[image]

I just realized that this is disabled by default for some reason I can't make sense of right now, and will investigate that, but if you enable it and reconnect, things should work just as expected.