OctoPrint/OctoPrint #4494: [Request] Send long filenames to Marlin when the WRITE_LFN capability is set.

nophead, 2022-04-22:
### Is your feature request related to a problem? Please describe.

The latest Marlin bugfix branch has support for writing SD files with long filenames but OctoPrint sends 8.3 names, so only files written to the card externally show long filenames in the listing. Files upload via OctoPrint always have short names.

### Describe the solution you'd like

As the description. Full support for long filenames seems to be the future, so the core OctoPrint should support it, either by recognising WRITE_LFN or having another comms option.

foosel, 2022-05-25:
> either by recognising `LFN_WRITE`

Is that a newly introduced capability flag that gets included in the `M115` response?

cp2004, 2022-05-25:
Yes: https://github.com/MarlinFirmware/Marlin/blob/54c9358964a71e28dcff0f5f64055b07333757c4/Marlin/src/gcode/host/M115.cpp#L163

foosel, 2022-05-25:
Thanks, that clarifies that.

foosel, 2022-11-09:
Implemented and ready for 1.9.0.

If `LFN_WRITE` capability is reported as supported, OctoPrint by default now will upload SD files to the printer under their full name instead of shortening it to DOS conventions.

The primary name as reported by the firmware (first entry in the M20 output) is still considered the main name and used for any file operations like selecting or deleting. So, if the firmware supports a DOS name here, that's also what OctoPrint will (continue to) use for any such operations.

nophead, 2022-11-09:
Thanks, I will look forward to 1.9 being released. Shame I can only run it on machine until Raspberry PIs become available again.

foosel, 2022-11-09:
I hope you are aware that it also works just fine on any old laptop or other SBC? There's also a nifty little script by @paukstelis that sets everything up for you against multiple Linux distributions: https://github.com/paukstelis/octoprint_install

nophead, 2022-11-09:
Yes but I have 5 printers with the original model B running old versions of OctoPrint. I have only one with an RPI0 W2 running the latest Octoprint. 

If I update the printers to another random SBC I would have to model it in NopSCADlib to make new mounting brackets and possibly replace the cameras. I can wait because the printers work fine with old versions of OctoPrint and don't need the SD card to print good quality.

paukstelis, 2022-11-09:
> Yes but I have 5 printers with the original model B running old versions of OctoPrint. I have only one with an RPI0 W2 running the latest Octoprint.
>
> If I update the printers to another random SBC I [...]

Keep in mind that you can also easily run multiple printers off of a single Pi or any other hardware. I have another script for that: https://github.com/paukstelis/octoprint_deploy

nophead, 2022-11-09:
Not the way I run them. I use the GPIO to control the power supply and the lights and use the serial port on the GPIO for comms and the camera port. There is only one camera port and one serial port per RPI.

No doubt a multicore RPI could run more printers but it would need to use the USB for comms and cameras and then there would be earth loop problems.

closed 2023-05-23 by foosel