OctoPrint/OctoPrint #4500: backup creating hidden files

Will-wastelander, 2022-04-25:
### The problem

Backup file names are starting with "."

This has been happening for some times now, and I am just now getting a chance to open a ticket for it. I have opened a ticket w/ @jneilliii for his backupscheduler, but he says that he pulls the name from appearance. Which is blank on my Prusa system, that is currently seeing this issue. It happens with my Enders, and BIQU BX at the very least.

2022-04-24 20:44:54,231 - octoprint.plugins.backup - INFO - Creating backup zip at /home/pi/.octoprint/data/backup/.octoprint-startup-20220424-204453.zip (excluded: -)...

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

plugins are disabled

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

OctoPi 1.0.0

### Printer model & used firmware incl. version

Prusa 3.11, Marlin

### Browser and version of browser, operating system running browser

Chrome, Windows 10 21H2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[backup files](https://media.discordapp.net/attachments/708230241285439518/967149136484925550/unknown.png
[octoprint-systeminfo-20220424205208.zip](https://github.com/OctoPrint/OctoPrint/files/8551500/octoprint-systeminfo-20220424205208.zip)
)

Will-wastelander, 2022-04-28:
I first thought it was an issue w/ backupscheduler plugin, so I opened the below ticket. But then I found in the logs that it shows that backup is the one that appears to be causing the issue.
https://github.com/jneilliii/OctoPrint-BackupScheduler/issues/24

Will-wastelander, 2022-04-28:
> > Did the issue persist even in safe mode?
> > I cannot test this issue in safe mode (state why below)
> > If you could not test in safe mode, please state why
> > plugins are disabled
> 
> The backup & restore is not disabled in safe mode, so it should still be usable to test it from there.

But backupscheduler is. Which is what creates the backups on startup, etc, correct ?

Will-wastelander, 2022-04-28:
2022-04-28 15:01:15,211 - octoprint.server - INFO - Goodbye!
2022-04-28 15:01:16,876 - octoprint.startup - INFO - ******************************************************************************
2022-04-28 15:01:16,878 - octoprint.startup - INFO - Starting OctoPrint 1.7.3
2022-04-28 15:01:16,878 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2022-04-28 15:01:16,878 - octoprint.startup - INFO - Reason for safe mode: setting in config.yaml
2022-04-28 15:01:16,879 - octoprint.startup - INFO - ******************************************************************************
2022-04-28 15:01:17,176 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-04-28 15:01:17,219 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-04-28 15:01:17,219 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-04-28 15:01:17,220 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-04-28 15:01:17,774 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2022-04-28 15:01:17,823 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2022-04-28 15:01:18,967 - octoprint.plugin.core - INFO - Found 43 plugin(s) providing 18 mixin implementations, 38 hook handlers
2022-04-28 15:01:19,159 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2022-04-28 15:01:19,165 - octoprint.server - INFO - Intermediary server started
2022-04-28 15:01:19,182 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2022-04-28 15:01:19,230 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2022-04-28 15:01:19,234 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2022-04-28 15:01:19,234 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2022-04-28 15:01:19,234 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2022-04-28 15:01:19,234 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2022-04-28 15:01:19,235 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2022-04-28 15:01:19,235 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2022-04-28 15:01:19,235 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2022-04-28 15:01:19,236 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2022-04-28 15:01:19,236 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2022-04-28 15:01:19,236 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2022-04-28 15:01:19,236 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2022-04-28 15:01:19,237 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_manage'), Need(method='role', value='plugin_pluginmanager_install')")
2022-04-28 15:01:19,237 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2022-04-28 15:01:19,237 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2022-04-28 15:01:19,237 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2022-04-28 15:01:19,849 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python3 -m pip" as command to invoke pip
2022-04-28 15:01:20,390 - octoprint.util.pip - INFO - Version of pip is 20.3.3
[... cut ...]
2022-04-28 15:01:38,758 - octoprint.server.util.flask - INFO - Passively logging in user will from ::ffff:192.168.x.139
2022-04-28 15:01:38,759 - octoprint.access.users - INFO - Logged in user: will
2022-04-28 15:01:40,370 - octoprint.server.util.sockjs - INFO - User will logged in on the socket from client ::ffff:192.168.x.139
2022-04-28 15:03:58,130 - octoprint.plugins.backup - INFO - Creating backup zip at /home/pi/.octoprint/data/backup/.octoprint-backup-20220428-150358.zip (excluded: -)...

Will-wastelander, 2022-07-02:
Is there any update on this issue ?

cp2004, 2022-07-07:
I have just looked into this. Apologies for the delay, I personally have not had any time over the last couple of months for development, and it has been on the list for someone to look at for a while.

Looking at the code to understand what is going on:

- Backups are first processed with a temporary file name, which deliberately has a dot in front of it. This is also the name that is logged to the octoprint.log file as well.
  https://github.com/OctoPrint/OctoPrint/blob/2e1ec42c5b7602be7bb57b1792598a851441c156/src/octoprint/plugins/backup/__init__.py#L967-L968
- Then once everything has been added to the zip file, it is moved to the final location, which is the filename *without* the dot.
  https://github.com/OctoPrint/OctoPrint/blob/2e1ec42c5b7602be7bb57b1792598a851441c156/src/octoprint/plugins/backup/__init__.py#L1078

So my only possible conclusion from that is that the final move is not working correctly, and you are left with the hidden file only. I'm not sure how to fix that, but it is an update explaining what the code is doing....

Will-wastelander, 2022-07-07:
@cp2004 I can send you my config.yaml, or make an image of my SD card, and provide that. This happens on multiple systems of mine.

Will-wastelander, 2022-07-07:
Correct, the renaming/move does not appear to complete. Could there be a check added to confirm if renaming completed, if not throw a failed error ?

Will-wastelander, 2022-07-07:
Awesome! I will upgrade tonight.

jneilliii, 2022-07-07:
Will, is your final backup destination on a network share or remote location outside of octoprint's default location local?

Will-wastelander, 2022-07-07:
> Will, is your final backup destination on a network share or remote location outside of octoprint's default location local?

@jneilliii I thought that might be the case with my original issue, but this is happening on my prusa setup also, which does not use network shared.

Will-wastelander, 2022-07-27:
Ok, I have been running the cp2004 build for a few weeks now.. Here is what I have for backups so far.

`55004 -rw-r--r--  1 pi pi 56323508 Jul 13 17:45  Prusa-i3-MK3S-backup-20220713-174502.zip
59016 -rw-r--r--  1 pi pi 60430644 Jul 20 00:01 'Prusa i3 MK3S-daily-20220720-000014.zip'
59016 -rw-r--r--  1 pi pi 60430854 Jul 21 00:00 'Prusa i3 MK3S-daily-20220721-000015.zip'
59016 -rw-r--r--  1 pi pi 60430862 Jul 22 00:01 'Prusa i3 MK3S-daily-20220722-000016.zip'
59016 -rw-r--r--  1 pi pi 60430868 Jul 23 00:01 'Prusa i3 MK3S-daily-20220723-000017.zip'
59016 -rw-r--r--  1 pi pi 60430783 Jul 24 00:01 'Prusa i3 MK3S-daily-20220724-000018.zip'
59016 -rw-r--r--  1 pi pi 60431118 Jul 25 00:01 'Prusa i3 MK3S-daily-20220725-000020.zip'
59016 -rw-r--r--  1 pi pi 60431112 Jul 26 00:01 'Prusa i3 MK3S-daily-20220726-000021.zip'
37560 -rw-r--r--  1 pi pi 38459090 Jul 12 20:16 '.Prusa i3 MK3S-startup-20220712-201500.zip'
45048 -rw-r--r--  1 pi pi 46125833 Jul 12 21:50 '.Prusa i3 MK3S-startup-20220712-214910.zip'
30532 -rw-r--r--  1 pi pi 31261361 Jul 13 17:09 '.Prusa i3 MK3S-startup-20220713-170840.zip'
59016 -rw-r--r--  1 pi pi 60430613 Jul 14 17:42 'Prusa i3 MK3S-startup-20220714-174116.zip'
59016 -rw-r--r--  1 pi pi 60430638 Jul 17 00:00 'Prusa i3 MK3S-weekly-20220717-000011.zip'
59016 -rw-r--r--  1 pi pi 60430794 Jul 24 00:01 'Prusa i3 MK3S-weekly-20220724-000019.zip'
`

jneilliii, 2022-07-27:
Has it always been isolated to startup backups from my plugin? Based on that sample set it seems to be, but not all of them.

Will-wastelander, 2022-07-27:
Jneiliii,

I will have to look at my other printers to confirm, as they haven't been
restarted in a few months..

Will-wastelander, 2022-07-27:
@jneilliii Here is one of my other printers.
`pi@Ender-3-octo:~ $ ll .octoprint/data/backup/
total 5216820
     4 drwxr-xr-x  2 pi pi      4096 Jul 26 00:00  .
     4 drwxr-xr-x 18 pi pi      4096 Jul 26 18:20  ..
 78660 -rw-r--r--  1 pi pi  80541206 Aug  6  2021  .E3v1-EZB-backup-20210806-233022.zip
106116 -rw-r--r--  1 pi pi 108659168 Jul 13 00:00 'E3v1 EZB-daily-20220713-000013.zip'
106136 -rw-r--r--  1 pi pi 108681802 Jul 14 00:00 'E3v1 EZB-daily-20220714-000014.zip'
106160 -rw-r--r--  1 pi pi 108704136 Jul 15 00:00 'E3v1 EZB-daily-20220715-000015.zip'
106180 -rw-r--r--  1 pi pi 108726541 Jul 16 00:00 'E3v1 EZB-daily-20220716-000016.zip'
106204 -rw-r--r--  1 pi pi 108748702 Jul 17 00:00 'E3v1 EZB-daily-20220717-000018.zip'
106228 -rw-r--r--  1 pi pi 108771325 Jul 18 00:00 'E3v1 EZB-daily-20220718-000019.zip'
106248 -rw-r--r--  1 pi pi 108793869 Jul 19 00:00 'E3v1 EZB-daily-20220719-000021.zip'
106268 -rw-r--r--  1 pi pi 108816476 Jul 20 00:00 'E3v1 EZB-daily-20220720-000022.zip'
106296 -rw-r--r--  1 pi pi 108839051 Jul 21 00:00 'E3v1 EZB-daily-20220721-000023.zip'
106312 -rw-r--r--  1 pi pi 108861676 Jul 22 00:00 'E3v1 EZB-daily-20220722-000024.zip'
106340 -rw-r--r--  1 pi pi 108884253 Jul 23 00:00 'E3v1 EZB-daily-20220723-000026.zip'
106360 -rw-r--r--  1 pi pi 108907308 Jul 24 00:00 'E3v1 EZB-daily-20220724-000027.zip'
106380 -rw-r--r--  1 pi pi 108930289 Jul 25 00:00 'E3v1 EZB-daily-20220725-000028.zip'
106404 -rw-r--r--  1 pi pi 108952836 Jul 26 00:00 'E3v1 EZB-daily-20220726-000030.zip'
 97560 -rw-r--r--  1 pi pi  99895651 Sep  1  2021 'E3v1 EZB-monthly-20210901-000048.zip'
103176 -rw-r--r--  1 pi pi 105646139 Oct  1  2021 'E3v1 EZB-monthly-20211001-000006.zip'
103492 -rw-r--r--  1 pi pi 105971329 Nov  1  2021 'E3v1 EZB-monthly-20211101-000004.zip'
103492 -rw-r--r--  1 pi pi 105971326 Dec  1  2021 'E3v1 EZB-monthly-20211201-000010.zip'
103496 -rw-r--r--  1 pi pi 105972112 Jan  1  2022 'E3v1 EZB-monthly-20220101-000007.zip'
103592 -rw-r--r--  1 pi pi 106073005 Feb  1 00:00 'E3v1 EZB-monthly-20220201-000030.zip'
104072 -rw-r--r--  1 pi pi 106561733 Mar  1 00:00 'E3v1 EZB-monthly-20220301-000029.zip'
104328 -rw-r--r--  1 pi pi 106827249 Apr  1 00:00 'E3v1 EZB-monthly-20220401-000011.zip'
104776 -rw-r--r--  1 pi pi 107284385 May  1 00:01 'E3v1 EZB-monthly-20220501-000100.zip'
105224 -rw-r--r--  1 pi pi 107743582 Jun  1 00:00 'E3v1 EZB-monthly-20220601-000028.zip'
105888 -rw-r--r--  1 pi pi 108424142 Jul  1 00:00 'E3v1 EZB-monthly-20220701-000024.zip'
 81172 -rw-r--r--  1 pi pi  83119550 Aug 20  2021 '.E3v1 EZB-startup-20210820-222425.zip'
105496 -rw-r--r--  1 pi pi 108025998 Sep 10  2021 'E3v1 EZB-startup-20210910-121212.zip'
103096 -rw-r--r--  1 pi pi 105566070 Sep 27  2021 'E3v1 EZB-startup-20210927-115808.zip'
104016 -rw-r--r--  1 pi pi 106507501 Feb 21 21:11 'E3v1 EZB-startup-20220221-211012.zip'
104280 -rw-r--r--  1 pi pi 106775330 Mar 10 10:47 'E3v1 EZB-startup-20220310-104542.zip'
105024 -rw-r--r--  1 pi pi 107538991 May 18 23:25 'E3v1 EZB-startup-20220518-232507.zip'
105024 -rw-r--r--  1 pi pi 107541355 May 18 23:52 'E3v1 EZB-startup-20220518-235045.zip'
105068 -rw-r--r--  1 pi pi 107583434 May 28 22:14 'E3v1 EZB-startup-20220528-221404.zip'
105152 -rw-r--r--  1 pi pi 107675510 May 28 23:28 'E3v1 EZB-startup-20220528-232734.zip'
105152 -rw-r--r--  1 pi pi 107675633 May 28 23:29 'E3v1 EZB-startup-20220528-232915.zip'
105220 -rw-r--r--  1 pi pi 107740316 May 31 20:33 'E3v1 EZB-startup-20220531-203233.zip'
105460 -rw-r--r--  1 pi pi 107988593 Jun 11 20:09 'E3v1 EZB-startup-20220611-200912.zip'
105656 -rw-r--r--  1 pi pi 108189661 Jun 20 17:01 'E3v1 EZB-startup-20220620-170100.zip'
105880 -rw-r--r--  1 pi pi 108419146 Jun 30 18:43 'E3v1 EZB-startup-20220630-184230.zip'
105920 -rw-r--r--  1 pi pi 108461844 Jul  2 15:48 'E3v1 EZB-startup-20220702-154718.zip'
106092 -rw-r--r--  1 pi pi 108633245 Jul 11 20:23 'E3v1 EZB-startup-20220711-202317.zip'
105312 -rw-r--r--  1 pi pi 107832663 Jun  5 00:01 'E3v1 EZB-weekly-20220605-000034.zip'
105468 -rw-r--r--  1 pi pi 107992184 Jun 12 00:00 'E3v1 EZB-weekly-20220612-000007.zip'
105620 -rw-r--r--  1 pi pi 108150277 Jun 19 00:00 'E3v1 EZB-weekly-20220619-000017.zip'
105780 -rw-r--r--  1 pi pi 108310615 Jun 26 00:00 'E3v1 EZB-weekly-20220626-000001.zip'
105932 -rw-r--r--  1 pi pi 108469767 Jul  3 00:00 'E3v1 EZB-weekly-20220703-000012.zip'
106052 -rw-r--r--  1 pi pi 108591370 Jul 10 00:00 'E3v1 EZB-weekly-20220710-000023.zip'
106204 -rw-r--r--  1 pi pi 108748704 Jul 17 00:00 'E3v1 EZB-weekly-20220717-000018.zip'
106360 -rw-r--r--  1 pi pi 108907309 Jul 24 00:00 'E3v1 EZB-weekly-20220724-000027.zip'
  6988 -rw-r--r--  1 pi pi   7152018 Jan 29  2021  .octoprint-backup-20210129-000016.zip
`

cp2004, 2022-09-04:
@Will-wastelander were you ever able to find the log files needed to further analyse this issue?

Will-wastelander, 2022-09-07:
> @Will-wastelander were you ever able to find the log files needed to further analyse this issue? I don't care what state they are in with other messages or how large they are, I can easily manipulate it to show what I would like. With nothing, we can't look into the issue further as no one else can reproduce it.

I'm in the process of moving. I'll check next time I'm with the printer.

cp2004, 2022-09-07:
No worries, thanks for the update.

Will-wastelander, 2022-10-30:
Confirmed that this is still happening.

max246, 2023-02-02:
You can see there is multiple waning about not space left

`OSError: [Errno 28] No space left on device`

Might need to check if there is space left before starting the backup