OctoPrint/OctoPrint #4503: Octoprint losing z-axis causing nozzle to drop onto build plate between prints

JamieMelamed, 2022-04-29:
### The problem

Between prints the nozzle drops onto build plate. Appears that z-axis is lost.  Works fine from SD Card without Octoprint. Issue started a few days ago. Only thing that changed was I upgraded my iMac to a new one running Monterey 12.3.1 with the new M1 chip

Setup ender 3v2 w/raspberry pi
Python 3.7.3 OctoPi 0.18.0

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.7.3 

### Operating system running OctoPrint

MacOS

### Printer model & used firmware incl. version

ender 3v2  v4.2.2

### Browser and version of browser, operating system running browser

Chrome is up to date Version 101.0.4951.41 (Official Build) (arm64)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220429211002.zip](https://github.com/OctoPrint/OctoPrint/files/8594473/octoprint-systeminfo-20220429211002.zip)

Starting dependency resolution...
packed_core.js?05a5d975:1262 ... dependency resolution done
packed_core.js?05a5d975:1285 Initial application setup done, connecting to server...
packed_core.js?05a5d975:1086 Connected to the server
packed_core.js?05a5d975:1092 Triggering connect on all view models
packed_core.js?05a5d975:1094 Server (re)connect processed
packed_core.js?05a5d975:1285 Finalizing application startup
packed_core.js?05a5d975:1266 Going to bind 39 view models...
packed_core.js?05a5d975:1278 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?05a5d975:781 Chrome Arc Fix disabled
packed_core.js?05a5d975:781 Chrome Arc Fix enabled
packed_core.js?05a5d975:1278 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?05a5d975:1278 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?05a5d975:1278 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_update since it does not exist
packed_core.js?05a5d975:1278 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate_settings since it does not exist
packed_core.js?05a5d975:216 User Jmelamed logged in
packed_core.js?05a5d975:1283 ... binding done
packed_core.js?05a5d975:1283 Application startup complete
36The resource <URL> was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropr [... cut ...]
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/#temp:1 The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.

cp2004, 2022-04-29:
I am going to guess that you have something configured in the gcode scripts to disable the steppers?

cp2004, 2022-05-16:
@JamieMelamed any reply here?

JamieMelamed, 2022-05-16:
I have not changed anything other than upgrading my Mac. Would that be a CURA issue then?

cp2004, 2022-05-16:
Either in OctoPrint or Cura, both have gcode scripts. The M18 or M84 commands disable the steppers.

JamieMelamed, 2022-05-16:
My End G-Code in CURA looks good. Unless I'm missing something.

G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

jneilliii, 2022-05-17:
Even if you aren't disabling the z stepper in end gcode I do think there is still timeouts in Marlin that will turn them off after a period of time. Double check you don't have any of the disable commands in OctoPrint's gcode scripts settings. I think by default the octopi image will have the on print cancel command filled in and it does have disable steppers in there.

cp2004, 2022-05-27:
Since this doesn't look like a bug in OctoPrint, as mentioned it is probably the timeouts or gcode scripts, I'll close this.

If you need further support with your printer/general setup, please get in touch on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org) where helpful people are around to sort things out.

closed 2022-05-27 by cp2004