OctoPrint/OctoPrint #4504: Deleting files

appleimperio, 2022-04-30:
### The problem

1. octoprint is installed but the upload folder is in a NFS share volume try SMB the same thing happened. I have 3 octoprints running on different pi two instances  use dockers and the other one octoprint image all pointing to the same octoprint/upload NFS share folder.
2. I make a backup of one of the printers checking the boxes exclude uploaded files (since all my files are in a NAS sever I don't need the backup) and the other box exclude timelapses ( I don't use timelapses)
3. when I restore the backup I lost all my files in the NAS share
[octoprint-systeminfo-20220430223548.zip](https://github.com/OctoPrint/OctoPrint/files/8597929/octoprint-systeminfo-20220430223548.zip)
.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

since I starting from a fresh install of octoprint and restoring a backup there is no use for safe mode

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

raspbian lite

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

jneilliii, 2022-05-01:
I think this is by design. The backup is more of a snapshot in time and when you do the restore it will remove the entire basedir, including uploads folder, etc. and then restore what's saved in the zip into a completely clean basedir folder.

appleimperio, 2022-05-01:
the only workaround I find is restore the backup without mapping the NAS share and when the backup is up, map the share and restart octoprint that way I don't lost the files, but I discover that a little late all my files are gone. Thank you

closed 2022-05-01 by appleimperio