OctoPrint/OctoPrint #4505: Timelapse get stuck in UnRendering when filename includes (%) or special character

NotExpectedYet, 2022-05-07:
### The problem

On version 1.8.0-RC5 

1. Complete a print job with timelapse capture enabled and a file that has a special character in the name... i.e. (%)
2. See the file complete rendering and hit the finished timelapse list.
3. See also the timelapse is never removed from the unrendered queue. 

On previous versions (tested 1.7.2) this would not render at all with no warning. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0-RC5

### Operating system running OctoPrint

Linux

### Printer model & used firmware incl. version

Virtual Printer

### Browser and version of browser, operating system running browser

Firefox, Chome, Linux Ubuntu

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]
[octoprint-systeminfo-20220507051517.zip](https://github.com/OctoPrint/OctoPrint/files/8644515/octoprint-systeminfo-20220507051517.zip)
[Thumbnail_2%_Checker.zip](https://github.com/OctoPrint/OctoPrint/files/8644518/Thumbnail_2._Checker.zip)

cp2004, 2022-05-07:
Related to the fix in #4366

cp2004, 2022-07-01:
Addressed in #4563

commit ae4426e4 (2022-06-30): 🐛 Fix handling of % in timelapse filenames (#4563)

Switched all replacement logic to {} style format strings, which means
we no longer need to escape % internally. Only escaping it in one single
place now, before pacing the capture file pattern to ffmpeg, as THAT
uses % style formatting.

Closes #4505
closed 2023-05-23 by foosel