OctoPrint/OctoPrint #4509: navbar error on 1.8.0-rc5

G-Pereira, 2022-05-17:
### The problem

1. Start the server
2. Acess the server on the browser
3. Browser throws an Internal Server Error
4. Server logs the following:
```
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 2077, in wsgi_app
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 1525, in full_dispatch_request
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 1523, in full_dispatch_request
  File "/usr/lib/python3.10/site-packages/flask/app.py", line 1509, in dispatch_request
  File "/usr/lib/python3.10/site-packages/octoprint/server/views.py", line 348, in index
  File "/usr/lib/python3.10/site-packages/octoprint/server/views.py", line 1278, in fetch_template_data
KeyError: 'navbar'
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0-rc5

### Operating system running OctoPrint

Linux

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

Firefox 100.0.1

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

G-Pereira, 2022-05-17:
I've included the octoprint's source as a systeminfo bundle just to prevent this issue being deleted, since it can't be generated because the interface can't be accessed.

foosel, 2022-05-17:
Wrong - the system info bundle can also be generated from the recovery screen and the command line, so please do so.

https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887

cp2004, 2022-06-06:
@G-Pereira did you get anywhere with trying to find the systeminfo bundle?

cp2004, 2022-06-16:
If this is still and issue and you can get some of the information to us, please let us know and we can reopen the issue. In the meantime, we'll close this issue as there's nothing actionable here.