OctoPrint/OctoPrint #4515: GCode Viewer crashes on some gcode files

JeroenDelcour, 2022-05-20:
### The problem

Since v1.8.0, OctoPrint crashes when trying to visualize some gcode files with the GCode Viewer. It shows "Downloading" in the status bar forever, and sometimes I get a popup saying I need to reload the interface.

[image]

Loading the file while not in the GCode Viewer also takes very long (20+ seconds) and causes a large spike in CPU usage (running on a Raspberry Pi 4). Memory usage seems normal, no large spikes.

[image]

What's even weirder is that it leaves the GCode file partially or fully erased (0kb size) after restarting.

I'm not sure if it's related, but it seems like some gcode files take considerably longer to visualize than in v1.7.3. A 3.5MB file took over 20 seconds, where another larger file took under a second. Take this with a grain of salt, since I haven't done a timed comparison with v1.7.3.

Printing the file seems to work fine (still printing at the moment).

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Creality Ender 3, Marlin firmware

### Browser and version of browser, operating system running browser

Firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

I'm having trouble debugging, because I don't see any relevant entries in octoprint.log. It seems to crash and restart before it gets a chance to log anything. There's nothing in the browser console, either.

I'm not sure why some gcode files work while others don't. I tried removing parts of a file that it crashes on to see if there was a particular line that's responsible. This particular gcode file doesn't crash if I remove the gcode for the final layer. I don't see anything unusual in those lines, just regular G1 and G0 commands.

I've attached the gcode file with everything except the last layer that works, as well as the full gcode file that causes the crash. I've also attached another gcode file sliced from a completely different STL that also causes a crash. The octoprint.log and systeminfo bundle is also there.

[octoprint crash files.zip](https://github.com/OctoPrint/OctoPrint/files/8738411/octoprint.crash.files.zip)

jneilliii, 2022-05-20:
Try in safe mode. I suspect the issue is with DisplayLayerProgress as it reacts to file select events.

JeroenDelcour, 2022-05-20:
I already tried it in safe mode, doesn't appear to make a difference.

cp2004, 2022-05-20:
> I'm not sure if it's related, but it seems like some gcode files take considerably longer to visualize than in v1.7.3. A 3.5MB file took over 20 seconds, where another larger file took under a second. Take this with a grain of salt, since I haven't done a timed comparison with v1.7.3.

That's unfortunate - we put a lot of effort into making the file loads much quicker and using less memory!

I don't like the sound of it erasing the files when it crashes though, the changes in the gcode viewer itself in 1.8.0 shouldn't have had an effect there as they're all client side, so there must be something else wrong.

Will have to test it out with the files provided and see - it will be interesting to know if it is specific to your file on your OctoPrint server or the actual file itself.

jneilliii, 2022-05-20:
I just tried the files and am unable to reproduce the issue. If you are attempting to select files that were previously uploaded with plugins enabled (not in safe mode) the file could still be corrupted in a way that doesn't allow selecting. Did you happen to try re-uploading the file while running in safe mode? 

[image]

circuitmike, 2022-05-20:
Not sure if this is related, but I just upgraded and I'm also having a problem with the gcode viewer. If I use the viewer while the printer is heating up, it shows a blank build plate with the print head in the home position (lower left when viewed from above, for my Prusa i3 MK3S+). Once the print starts, it'll display the gcode correctly. I'm not sure what's going on, but the gcode viewer definitely seems to have some new bugs!

cp2004, 2022-05-20:
> Not sure if this is related, but I just upgraded and I'm also having a problem with the gcode viewer. If I use the viewer while the printer is heating up, it shows a blank build plate with the print [...]

Do you have it following the print job? That would be expected I think if 'Sync with job progress' is checked. No gcode (actual extrusion) is sent to the printer yet so nothing is shown?

circuitmike, 2022-05-20:
I do have it following the print job, yes. But I also have the "Also show current layer" feature turned on, so I would expect it to show the current layer in a lighter color, not show nothing at all. I thought that's how it worked in the last 1.7 version (1.7.3?).

JeroenDelcour, 2022-05-21:
Turns out it was a power supply issue of my own doing!

I was having trouble recreating the issue. It would only sometimes crash, and on different gcode files. On a few occasions I saw an undervoltage warning. Monitoring an SSH connection and looking at the uptime revealed that OctoPrint wasn't crashing, the whole Raspberry Pi 4 was rebooting.

I swapped the 1A power supply for a 3A one (which is actually the minimum requirement for a Pi 4, my bad!). That seems to have solved the issue. I suppose this also explains why the files were getting corrupted.

Sorry for wasting everyone's time!

However, loading some files into the GCode Viewer still seems to take a long time. The same 3.5MB file (attached below) still takes just over 20 seconds (in safe mode, though unfortunately I don't have the gcode on my computer anymore so I can't try @jneilliii 's suggestion of also uploading it in safe mode). Most of that time is spent parsing, according to the progress bar. There's no spike in CPU usage anymore, though. Other (larger) files are parsed much quicker - I don't see an immediately obvious pattern. Would you like me to make a separate issue for this, or should I first check that it was faster in v1.7.3?

[21-297Balljoint8mm.zip](https://github.com/OctoPrint/OctoPrint/files/8747313/21-297Balljoint8mm.zip)

closed 2022-05-21 by JeroenDelcour

cp2004, 2022-05-21:
That gcode file does take longer than another of equivalent size that I have here - not investigated why. I did time some benchmarks compared to 1.7.3:

*All files were already cached by the browser, so no download times were included (all 304 responses), immediately after each other*

| Version | Your file (3.5MB)  | My file (3.0MB) |
|---------|--------------------|-----------------|
| 1.8.0   | 4.2secs            | 1.4secs         |
| 1.7.3   | 5.2secs            | 2.4secs         |

*Microsoft Edge, Intel i7-1165G7*

So there is a notable improvement from 1.7.3 => 1.8.0 still, *and* the fact that your gcode file just takes longer to parse.

Maybe it is the type of gcode - there's lots of smaller lines in the circles, compared to mine that is longer, squarer shapes? Not sure. Maybe we will discover more performance issues in the future, but unless it's confirmed as being slower I won't go investigating too much.

JeroenDelcour, 2022-05-22:
Very nice benchmarks! Thanks for doing those.

I'm glad to see v1.8.0 is still an improvement over v1.7.3. Though it is interesting that files of similar size can take drastically different times to parse, there's no need to investigate further for my sake. I'm perfectly happy with the current performance :)