OctoPrint/OctoPrint #4519: Github access token not being saved

bruvv, 2022-05-23:
### The problem

When updating to 1.8.0 I got rate limited by github. I added my own access token from github that I generated and added it to octoprint 1.8.0 acording to: https://community.octoprint.org/t/i-get-something-about-rate-limit-exceeded-when-trying-to-update-octoprint-or-plugins-help/33550 but when saving the github access token isn't saved and stays empty after a reboot.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0

### Operating system running OctoPrint

docker

### Browser and version of browser, operating system running browser

Firefox latest version

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220523152453.zip](https://github.com/OctoPrint/OctoPrint/files/8754880/octoprint-systeminfo-20220523152453.zip)

Firefox console:
`ConfiguredLoggers has not changed. Not saving.`

cp2004, 2022-05-23:
It is designed to be set-only, for security reasons it is not exposed to be read again on the API. You can check directly in config.yaml that it is saved there.

bruvv, 2022-05-23:
Ah got it, we should then get some feedback in the GUI if there is token imho.

foosel, 2022-05-23:
> we should then get some feedback in the GUI if there is token imho

Agreed, can you open a feature request please, @bruvv?

Closing this bug report as everything is indeed working as intended (as far as I understood the last comment of you).