OctoPrint/OctoPrint #4522: serial not accessible in octoprint image

rlfnb, 2022-05-24:
### The problem

Got a fresh image with octoprint to update to python3. After booting, I was not able to connect the serial to my previously working prusa i3mk. The /dev/ttyAMA0 was already there, everything's fine so far... But octoprint could not open/use the serial. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Prusa i3mk

### Browser and version of browser, operating system running browser

Chromium

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Problem was solved by adding `disable-bt` to the end of of config.txt.

This change solved the resource clash between BT and serial (same name is used for reasons...)

Next change was to make sure the login shell is not bound to the serial: changed it in raspi-config, but you might find a better place directly in the image?

rlfnb, 2022-05-24:
its octopi related, so the system info bundle wont work here? you can easily reproduce the name clash according to: 
https://raspberrypi.stackexchange.com/questions/45570/how-do-i-make-serial-work-on-the-raspberry-pi3-pizerow-pi4-or-later-models/45571#45571

foosel, 2022-05-24:
I take it you have your printer not connected through USB but rather through the Einsy header? That's what `ttyAMA0` indicates, and in this case this behaviour is normal (and also not anything that Octo**Print** could change anything about, so wrong repository) and the solution is already found in Prusa's setup guide for OctoPi for using the Einsy port with OctoPrint: https://help.prusa3d.com/article/octoprint-configuration-and-install_2182.

closed 2022-05-24 by foosel

rlfnb, 2022-05-24:
as octopi is already a single purpose OS for OctoPrint it might help others to have disable-bt already in the image? anyway, thanks for your reply!

foosel, 2022-05-24:
It might also cause issues for others because now Bluetooth doesn't work anymore for them 🤷‍♀️ 

OctoPi (which, again this is *NOT* the repository of) has always tried to stay as close to stock Raspberry Pi OS, shipping with a default overlay that outright disables hardware for everyone would severely divert from that.

rlfnb, 2022-05-24:
ok, thank you for the explanation. Maybe a FAQ entry will help others?

foosel, 2022-05-25:
Added to https://community.octoprint.org/t/octoprint-wont-connect-to-my-prusa-mk3-mk3s/13532