OctoPrint/OctoPrint #4523: This version of the Python environment (3.7.5) is not supported for direct updates.

elpeter25, 2022-05-25:
### The problem

For the 1.8.1 version Python 3.7.5 I just installed for upgrading to 1.8.0 is not enough. Well I had to do a fresh install for the 1.8.0 with python 3.7 as I wasn't able to upgrade in any other way. I have a custom set up. Ubuntu on a NUC running multiple instances of octoprint.
Today I receive the following error:

This version of the Python environment is not supported for direct updates.

And I'm using latest Python 3.7 for ubuntu so... how is this even possible?

See screenshot bellow:
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0

### Operating system running OctoPrint

Ubuntu 18.04.6 LTS

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220525082226.zip](https://github.com/OctoPrint/OctoPrint/files/8768437/octoprint-systeminfo-20220525082226.zip)

cp2004, 2022-05-25:
When you open the software update settings, it should give you more details of what exactly is not up-to-date enough. It is not necessarily Python, it could be pip or setuptools as well.

elpeter25, 2022-05-25:
Thanks for that, didn't know it :)

Now it's even better :P
[image]

elpeter25, 2022-05-25:
Voilá! It worked :)

Thanks for pointing that out for me!!
[image]

cp2004, 2022-05-25:
👍

closed 2022-05-25 by cp2004