OctoPrint/OctoPrint #4524: [Request] Move the systeminfo txt data to it's own semi-hidden view under a URL or query parameter

foosel, 2022-05-25:
### Is your feature request related to a problem? Please describe.

People still keep sharing the txt data instead of the zip. Even if they have to click once more to get it.

Example: https://community.octoprint.org/t/hot-end-crashes-into-bed-when-octoprint-connects/44599

### Describe the solution you'd like

Remove the abridge system info txt data from the about dialog altogether. Instead, have it hide either between a query parameter or a dedicated (admin only) view.

### Describe alternatives you've considered

Sounding like a broken record constantly reminding people that no, we want the zip. That's no fun for me or any of the very helpful folk running after the bundle on the forums and such.

deepsol-oba, 2022-05-27:
Would you want to consider removing the "More" toggle altogether and instead suggest in the text to review data shared by unzipping the bundle downloaded before uploading?

My thinking is that the information shown in the "More" section is misleading as the zip file contains more information/details than shown in the section.

A later backlog item could be suggested to upload the zip file directly from the UI (where (https://bundleviewer.octoprint.org/) can be accessed by the Octoprint instance - possibly with showing the full set of information shared before uploading.

foosel, 2022-05-30:
> Would you want to consider removing the "More" toggle altogether and instead suggest in the text to review data shared by unzipping the bundle downloaded before uploading?

I can already see people unzipping and the rezipping it wrong so the bundleviewer no longer works 😅

The current panel already tells people that they can check the contents by going to the bundleviewer:

[image]

> A later backlog item could be suggested to upload the zip file directly from the UI (where (https://bundleviewer.octoprint.org/) can be accessed by the Octoprint instance - possibly with showing the full set of information shared before uploading.

That is something I'd like to avoid. Storage costs money, and can be abused. The bundleviewer in its current iteration is a strictly client-side only tool that doesn't require any backend. 

The issue also doesn't seem to be that people are worried about information that might be contained in the bundle, but rather that they are finding it easier or more intuitive to share something via copy and paste than in shape of a zip file that they believe someone then has to open (which isn't the case, but apparently the tooling available here in shape of the bundleviewer gets overlooked).

The idea is to make this `More` content vanish for the average user altogether, and only display it if certain query parameters are provided, or on a special view (similar to the recovery page). Removing it COMPLETELY would be kinda bad since it is a valuable quick debugging tool, but it is apparently too easily accessible now, meaning people rather share that instead (highlight, copy, paste) instead of the zip (click, download, upload).

foosel, 2022-11-07:
Ready for 1.9.0

foosel, 2023-05-23:
1.9.0 has been released

closed 2023-05-23 by foosel