OctoPrint/OctoPrint #4525: Fix double message when trying to skip a mandatory wizard during

foosel, 2022-05-25:
### Problem

  - run setup wizard (or trigger some other mandatory wizard steps to show up)
  - try to skip a mandatory steps by clicking "Next" without filling it in
  - Result: You get a warning that the step is mandatory *twice*

### Solution 

Find the reason for that and fix it.

max246, 2022-06-27:
I have investigated a little bit on this issue, apparently on plugin bootstrapWizard is binding the click event twice in someone to the button NEXT and PREVIOUS.

I am not familar with this plugin but seems like the method "onTabShow" and "onTabChange" is getting called twice as soon as you click the button.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel