OctoPrint/OctoPrint #4526: [Request] Bundle the Custom Control Editor Plugin

foosel, 2022-05-25:
### Is your feature request related to a problem? Please describe.

The Custom Control Editor plugin offers a UI based way to edit custom controls. It's been created and maintained by @Salandora but they no longer have time for it and it can be considered abandoned now (see also OctoPrint/plugins.octoprint.org#982). It is in need of maintenance.

### Describe the solution you'd like

Bundle the plugin into OctoPrint itself, but on an external release cycle to allow for easier maintenance support by the community.

### Describe alternatives you've considered

  * Losing the functionality. Not a good idea given its popularity.
  * Building an alternative right into OctoPrint. Not a good idea given the work has already been done.
  * Putting it straight into OctoPrint's source tree. Not a good idea since that would force it on the same release cycle.
  * Finding a new maintainer for it and keeping it unbundled. Would certainly be an option but I'm leaning towards this functionality meriting inclusion into the core package, given that it's repeatedly been confused with core functionality. 

Thoughts?

cp2004, 2022-05-25:
I looked into it a bit a while back, and decided if it were up to me *personally* I wouldn't use the right-click context menu so much (or maybe at all) - it's confused people before that there's just a green box when you first install it. It's not a design pattern that's commonly used. At the time I felt like it needed a bit more to make it clear to people how it works.

I've (very) tentatively got working on this on my list for the summer - so July onwards... But that's only if no one else works on it before then!

jneilliii, 2022-05-25:
Yeah, I would implement something like what was done in event manager rather than the right-click thing. What I ran into with some of my plugins and this approach with buttons was how to display it to the end user as if it were live in the control tab as they add it. Terminal Commands Extended is close, but even it has it's quirks. There are some hidden gems with custom controls though that I never integrated. 

[image]

cp2004, 2022-05-25:
There's definitely merit to having the WYSIWYG-style editor, so maybe finding a way to convey the actions alongside that would be good. I have high standards in my imagination 😅

jneilliii, 2022-05-25:
For my implementation it's a click the button to open a properties modal kind of solution. The other bits are just for moving/duplicating/deleting.

foosel, 2022-05-26:
Considering that my focus will rather be on knowledge transfer into the docs and similar bus factor aversion strategies, I'm pretty sure that at least I won't be working on this during the summer, so by all means, I'm open to alternative UI approaches. Anything that improves the UX *and* reduces my workload is welcome 😬

foosel, 2024-12-02:
The functionality is now provided by a new bundled plugin called "Custom Control Manager". There were a lot of UX changes. Got rid of the context menu and instead added some overlay controls, also made sure that it's clearer if no custom controls are configured yet.

https://github.com/user-attachments/assets/ffc12fe5-6aa9-4a9c-b2a0-ea37208309c9

[image]

There's no backend on the plugin, it's really just a frontend and utilizes the existing APIs. The core implementation of the custom controls feature has been slightly changed to now utilize the settings API instead of a custom control specific endpoint, and the settings API has been extended to allow access to custom control definitions (given the needed permissions). That should make everything a bit more aligned.

Will ship with 1.11.0.

closed 2025-04-22 by foosel