OctoPrint/OctoPrint #4527: [Request] Add a helper component to the SoftwareUpdate plugin to mark the version cache as stale for certain components

foosel, 2022-05-25:
### Is your feature request related to a problem? Please describe.

A third party plugin might want to be able to mark its version cache used for checking for updates as stale.

IIRC @jneilliii expressed such a wish (if so, could you please elaborate again?).

### Describe the solution you'd like

Add a helper component to the Software Update plugin that allows marking version cache contents as stale by component id.

jneilliii, 2022-05-25:
IIRC the issue I was experiencing was downgrading when switching release channels wouldn't happen.

foosel, 2022-05-25:
Hm, I thought we tackled that with #4238?

foosel, 2022-05-25:
Hm, then I'm going to unschedule this from 1.9.0 again until someone can remember why that was still needed 😅 I was only going through my Todoist tasks for OctoPrint and doing some initial cleanup of stuff that was in there but rather belonged on the issue tracker 😬