OctoPrint/OctoPrint #4528: Missing python modules after update to 1.8.0 docker container

RoboMagus, 2022-05-25:
### The problem

I've got an automatic update scheduled for most of my docker containers, of which Octoprint is one.
After the update to 1.8.0 I noticed a number of plugins were no longer functioning. Turns out that these rely on python module `six` which used to be packaged in the docker image.
The fix was easy enough by just running `pip install six` from within the container, but it this feels like it should be noted as a bug nonetheless.

If not _technically_ a bug then at least it's a heads-up.
(Perhaps should change it to a FR, to check for plugin dependencies after an Octoprint update and install any that are missing).

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Well... Its about dependencies for 3rd party plugins that are no longer present after an update... Safemode kind of beats the purpose.

### Version of OctoPrint

1.8.0

### Operating system running OctoPrint

Linux (Docker Image)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Errors in log due to missing module: 
```
2022-05-25 18:27:12,297 - octoprint.server - INFO - Goodbye!
2022-05-25 18:27:20,666 - octoprint.startup - INFO - ******************************************************************************
2022-05-25 18:27:20,670 - octoprint.startup - INFO - Starting OctoPrint 1.8.0
2022-05-25 18:27:20,671 - octoprint.startup - INFO - ******************************************************************************
2022-05-25 18:27:21,259 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-05-25 18:27:21,314 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-05-25 18:27:21,314 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-05-25 18:27:21,314 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-05-25 18:27:21,505 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2022-05-25 18:27:21,519 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python3.8/site-packages/octoprint/plugins, /octoprint/octoprint/plugins and installed plugin packages...
2022-05-25 18:27:22,029 - octoprint.plugin.core - INFO - Plugin Virtual Printer is disabled.
2022-05-25 18:27:23,718 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.3.1) is disabled.
2022-05-25 18:27:23,756 - octoprint.plugin.core - ERROR - Error loading plugin mqtt
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 1274, in _import_plugin
    module = _load_module(module_name, spec)
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_mqtt/__init__.py", line 5, in <module>
    import six
ModuleNotFoundError: No module named 'six'
2022-05-25 18:27:24,222 - octoprint.plugin.core - ERROR - Error loading plugin octolapse
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 1274, in _import_plugin
    module = _load_module(module_name, spec)
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_octolapse/__init__.py", line 29, in <module>
    import octoprint_octolapse.log as log
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_octolapse/log.py", line 27, in <module>
    import six
ModuleNotFoundError: No module named 'six'
2022-05-25 18:27:24,264 - octoprint.plugin.core - ERROR - Error loading plugin DryRun
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 1274, in _import_plugin
    module = _load_module(module_name, spec)
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_DryRun/__init__.py", line 9, in <module>
    from octoprint_DryRun.gcode_parser import Commands, ParsedCommand, Response
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_DryRun/gcode_parser.py", line 25, in <module>
    from six import string_types
ModuleNotFoundError: No module named 'six'
2022-05-25 18:27:24,714 - octoprint.plugin.core - INFO - Found 45 plugin(s) providing 43 mixin implementations, 85 hook handlers
2022-05-25 18:27:24,935 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2022-05-25 18:27:24,971 - octoprint.server - DEBUG - Starting intermediary server on http://0.0.0.0:5000
2022-05-25 18:27:24,976 - octoprint.server - INFO - Intermediary server started

...

```

SystemInfo Bundle:
[octoprint-systeminfo-20220525204347.zip](https://github.com/OctoPrint/OctoPrint/files/8773809/octoprint-systeminfo-20220525204347.zip)

jneilliii, 2022-05-25:
Seems more of an issue with a plugin not specifying it's requirements properly assuming six was already there for OctoPrint?

cp2004, 2022-05-25:
I can't see `six` as an explicit dependency in 1.7.3 either. Maybe it was a dependency-of-a-dependency, and since they have been updated there's no longer a requirement for it to be installed when building a new environment. It is a Python 2/3 compatibility library, so I'd say it's *very* likely that since most dependencies are also Py3 only now they've dropped using it.

Explicit plugin dependencies are always checked by pip, but if they aren't listed in the plugin then it won't install them of course. My guess is that this isn't the only dependency that has 'just worked' when it was imported because something else specifically required it.

RoboMagus, 2022-05-26:
It's probably true that any plugin using `six` should have at least listed it as part of its dependencies, even [Octoprint-MQTT](https://github.com/OctoPrint/OctoPrint-MQTT) fell because of this. 

@cp2004, Indeed I could also not find any trace of `six` being explicitly installed, but i figured an update of Octoprint should never actually break working plugins.

As for solving it on the plugin side by listing the dependency propperly, will a plugin update rather than a fresh install also pick up new dependencies from `setup.py`?

cp2004, 2022-05-27:
> but i figured an update of Octoprint should never actually break working plugins.

The way most python environments work is that they don't actually uninstall packages that are no longer needed, so they just stick around. So for most people they won't even notice, until they do a completely clean install again. In your case it seems that your update process is actually a fresh environment as well (beyond just the usual pip install).

We can't guarantee that nothing will break, especially not (probably) dependencies of dependencies like this. Not feasible to account for literally every possibility. There will inevitably be some changes required by plugins every now and then to adapt.

> As for solving it on the plugin side by listing the dependency propperly, will a plugin update rather than a fresh install also pick up new dependencies from setup.py?

Yes, to install the new package the process is the same - run setup.py, and if it has dependencies that are unmet they will be installed.

If you have the specific plugins that have this issue, then it should be reported directly to the plugin author. They can then decide what to do - whether to include the package or just not support Python 2 entirely - as nearly 20% of plugins have done.

closed 2022-05-27 by cp2004

RoboMagus, 2022-05-27:
> In your case it seems that your update process is actually a fresh environment as well (beyond just the usual pip install).

Well, thats just what the official Octoprint docker image does. Nothing special that I've done with it.

Come to think of it, even _if_ all of these plugin authors had listed `six` as a dependency, it would still have been an issue. As it would have been there (In the docker image) upon initial installation of the plugin. Then after the upgrade to 1.8.0, that package would no longer be found. As `setup.py` is not called again when the container starts it would throw the same exceptions as I've shown above.

Yes, to install the new package the process is the same - run setup.py, and if it has dependencies that are unmet they will be installed.

That's not what I asked. I want to know if a __Plugin Update__ (i.e. through the `software update` of the Octoprint UI) would pick up newly added dependencies listed in `setup.py`.

cp2004, 2022-05-27:
> Come to think of it, even if all of these plugin authors had listed six as a dependency, it would still have been an issue. As it would have been there (In the docker image) upon initial installation of the plugin. Then after the upgrade to 1.8.0, that package would no longer be found.

No, it would not be an issue. In order for the plugin to be installed it would have to have its dependencies however you set it up. In the official docker container these should be persisted along with the plugin packages themselves - if plugins & their were not persisted then you would have far far more problems than just this single dependency.

> > Yes, to install the new package the process is the same - run setup.py, and if it has dependencies that are unmet they will be installed.

> That's not what I asked. I want to know if a Plugin Update (i.e. through the software update of the Octoprint UI) would pick up newly added dependencies listed in setup.py.

The answer is the same, apologies if that is not clear. Installing an 'update' to a package is the same as installing it for the first time - it downloads the new version and installs it over the top of the old one.

jneilliii, 2022-05-31:
The closest I could find was the removal in https://github.com/OctoPrint/OctoPrint/commit/e00375e58654dbe881d217fa8452810de732119d and when it was added in https://github.com/OctoPrint/OctoPrint/commit/39cda08d0705cd8de98b7349bdd7c9e8b586bd0b

foosel, 2022-05-31:
Scanning if all dependencies are still satisfied after a docker image update is something that would have to be done on the docker side of things though. OctoPrint itself does not and will not expect that half of its venv suddenly gets removed while plugins still stay partially installed somehow. I honestly don't even know how the Docker image even accomplishes that, it sounds like a volume mount in the site packages and that's a recipe for trouble for precisely that reason.

I suggest to open a feature request on the OctoPrint/docker repository or get in touch with the maintainers on the forums or Discord about this, as it sounds like there's a fundamental issue present here.

Again though, that is down to how the Docker image specificallv does things, and checking all plugins on consistency on every start and reinstalling them from scratch if they aren't is WAY outside of the scope of OctoPrint itself. The whole reason why I chose to go with Python's own packaging infrastructure for plugin installation and dependency management was precisely because I did not want to reinvent that particular wheel.