OctoPrint/OctoPrint #4539: Server will not start when network has IPv6 connectivity (with some issues probably)

gabrieled07, 2022-06-10:
### The problem

I've come across an issue with octoprint with something not directly related. 
I have an Octoprint instance (1.8.1 -now-) installed manually on a Debian 10 host on an standard pc (x86_64) running on all up-to-date versions OS and python3.
(its been running flawlessly for +2 years, amazing software!!)

recently, the network I'm using with this host had implemented IPv6 dual stack (maybe with some minor dns issues, still early deployment)

When Debian started, got a valid, (and working) ipv6 address and OctoPrint did not start
after a quick check, there was no tcp bind. At this point I was unsure if last time I've updated and broke it, (yes, i've readed something about python and ignore it on the announcements for quite some time... sorry!) so, updated everything, python3 specifically and reinstalled octoprint with last 1.8.1.
Octoprint still was not binding/starting

log:
pre update:
2022-06-10 09:19:28,860 - octoprint.startup - INFO - ******************************************************************************
2022-06-10 09:19:28,864 - octoprint.startup - INFO - Starting OctoPrint 1.7.0
2022-06-10 09:19:28,865 - octoprint.startup - INFO - ******************************************************************************
2022-06-10 09:19:28,943 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-06-10 09:19:28,958 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-06-10 09:19:28,959 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 192.168.20.1:53 is working
2022-06-10 09:19:28,959 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working

post update:
2022-06-10 11:15:15,423 - octoprint.startup - INFO - ******************************************************************************
2022-06-10 11:15:15,426 - octoprint.startup - INFO - Starting OctoPrint 1.8.1
2022-06-10 11:15:15,428 - octoprint.startup - INFO - ******************************************************************************
2022-06-10 11:15:16,732 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-06-10 11:15:16,749 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-06-10 11:15:16,749 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 192.168.20.1:53 is working
2022-06-10 11:15:16,750 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working

at this point octoprint does nothing, it will not give any errors and will not continue starting (left 1.7.0 'starting' last night until today)

I suspected something with ipv6, but the host and the name resolution worked at that point (you can resolve and ping octoprint.org)
after a little of troubleshooting I've disabled ipv6 (setup manual invalid ipv6 address with no gateway) and OctoPrint worked as expected

Conclusions:
- Something failed at the connectivity check stage and prevented OctoPrint to continue starting
- Whatever it failed, was not a timeout and generated a condition where the server did not knew what to do
- No error message was logged or generated, neither crashed or stopped, making the troubleshooting a little bit difficult
- The issue can be replicated on 1.8.1 as well on 1.7.0
- Re-enabling ipv6 connectivity generate the same issue again

Thanks for your time, and again... amazing work!!

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

Debian 10

### Printer model & used firmware incl. version

home made with marlin

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220610122345.zip](https://github.com/OctoPrint/OctoPrint/files/8880162/octoprint-systeminfo-20220610122345.zip)

gabrieled07, 2022-06-21:
Hi Gina! sorry about the delay....
Both stacks were running at the same time, both with valid and working addresses (valid and working: can route to remote network 4 & 6, can resolve both via ipv4 and 6 ie nslookup -> server aaaa & a, and can download something from each one, ie. wget aaaa.host.net/file | a.host.net/file)
The topology its quite simple, internal network is served by dhcps on each stack, on wan side I have a /64 prefix delegation through dhcp-client. V4 via pppoe. Both stacks are running in parallel, no tunnels or translations.

About your comments, it is possible, OP did report that the connectivity check was ok, so maybe something happened when it tries to bind a socket to both 4 & 6? or to discover them?
If there is any way I can make the start more verbose (sorry I'm not super familiar with OP) I'll continue a little more of troubleshooting. 
Please tell me if you need me to check or test something else

gabrieled07, 2022-07-07:
Hi Gina, and sorry again about the delay, it was a very busy couple of days at work.
OP was left off all this time, and today I had time to continue with this issue.

In the mean time, the ipv6 implementation was kinda deployed (I'm involved in that)

So, when I was ready to start debugging the server, it started without problems, it binds to -4, to -6 and to both as expected, and works as expected.

At this point, I'm sure we had some recursive issues with our DNS that may caused this problem, however I cant replicate it just yet. I will be working on this the next days, probably deploying a new *buggy like before* name server.

About your comment, you are right, and I can totally understand why a connectivity check exist as you envision this software, but I don't like that a lot, (things talking to the internet freely, I tend to use them behind vpns) sure you guess it by now... 

I'll keep you posted

Gabriel

foosel, 2022-08-01:
I'm back from vacation (hence the delay on my part) and a after reading this comment now a bit unclear on whether the problem even still persists now or not. Have you had a chance to cross check and can you clarify, @gabrieled07?

gabrieled07, 2022-08-31:
Hi @foosel, hope you had a nice time off.
After another small delay on my part I got some updates.

Replicating the issue was hard, the problem was an extra 'asimetrical' jump on some core ipv6 routers "on my IPS side" that caused this issue (nothing to blame to octoprint)

And, I think I might found the issue

With this condition, data kinda flows between tcp connections, but many never completes, or only get a few bytes....

When octoprint starts, it checks quite some things outside, among them:
https://plugins.octoprint.org/blacklist.json

my theory is, connection was successfully made, but never completed the transfer, so the process was hang, due to the get function doesnt have a default timeout (I don't know python that well)

https://github.com/OctoPrint/OctoPrint/blob/f430257d7072a83692fc2392c683ed8c97ae47b6/src/octoprint/server/api/settings.py#L326-L330

maybe adding `timeout=30` or something like this will prevent weird networking issues halting the server start or other processes?

What do you think Gina?

Regards,
Gabriel

hunsra, 2022-11-01:
I have the same issue with OctoPrint 1.8.6 (OctoPi 0.18.0) running on a Raspberry Pi 3B+.  I also have PiHole running in my network environment, which may complicate/contribute to the issue.  When I disable IPv6 on the Raspberry Pi, the OctoPrint startup issue disappears.  PiHole is configured to act as DNS for both IPv4 and IPv6.  I'm happy to provide any logs and configuration information if helpful.

foosel, 2022-11-02:
Sorry, I haven't had time yet to further look into this, the 1.8.x maintenance took all available debugging and development time the past three months, and now I really need to get around to some of the PRs that have been sitting in the review queue for ages now. 

In general, any detective work from people seeing these issues would help a lot.

What I can tell y'all is that all outgoing network requests *should* have a timeout of 3.1s. It would be interesting to know if it's only the blacklist connection that is causing issues, by disabling the blacklist processing via the settings.

foosel, 2022-11-17:
Yeah.. so... turns out a timeout was indeed *not* set. The request had a timeout value from the fetch method, the fetch method had a timeout value from the startup call, the startup call fetched the timeout from the config, and the default value in the config was... non existing and thus there was no timeout. Hooray.

You know, it's really a good idea to first check that your assumptions are actually correct (here: the timeout is set) before you spend ages on trying to figure out why it doesn't get applied correctly 🙈

So, the current guess on what happens here is the following:

- OctoPrint tries to fetch the blacklist
- something about the dual stack config confuses stuff and the request gets stuck
- server startup hangs

So @gabrieled07 was spot on, just not with the code location. 

What I'm going to do now is making sure there actually IS a timeout being applied and that should solve the hanging. What it won't solve is the network configuration causing requests to not get proper responses. I have no idea what is happening there, but I could imagine some weird data flow issues possibly caused by firewalling with connection tracking (rather common practice) or some routing problems. Definitely out of scope for OctoPrint to solve, but the server should not be hanging when such an issue is encountered, and that's what I'm going to tackle here.

foosel, 2022-11-17:
Fix ready for 1.9.0

gabrieled07, 2022-11-22:
Great you could find and fix the issue! and I'm glad I could help a little bit to make Octoprint better!

commit 4b7085e3 (2022-11-17): 🐛 Fix missing timeout for blacklist fetch

Closes #4539
closed 2022-11-22 by gabrieled07