OctoPrint/OctoPrint #4543: Octoprint unnecessarily limits temperature offsets to +-50 degrees.

richfelker, 2022-06-16:
### The problem

1. I was trying to set a temperature offset of 70 degrees to print a gcode file sliced for 210 nozzle temperature with a material requiring 280 nozzle temperature.
2. Offset dialog textbox turned red and popped up a message that offset needs to be between -50 and 50, and would not let me submit.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.0

### Operating system running OctoPrint

OctoPi 0.18.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

There is no reason for arbitrary limits here. The range of temperatures hotend can vary by is well over 150 degrees, not 50 degrees, and limiting it just forces users to gratuitously reslice or hand-edit gcode files.

richfelker, 2022-06-16:
System info bundle is not relevant to this bug report.

foosel, 2022-06-17:
A bug report always requires a system info bundle, even if just to confirm you are indeed running the versions you think you are running, that you tested safe mode and so on. It takes two clicks to download and provide here, it cost you more time to claim it isn't relevant than it would have taken you to upload it. 

With that being said, the offset limitation is intentional to prevent people from shooting themselves in the foot. I'd be open to a feature request to make them configurable however.