OctoPrint/OctoPrint #4548: [Task] Replace manual path globbing & iterating with pathlib usage?

foosel, 2022-06-21:
We are currently using `os.scandir`. With Python 3 now in use we could switch this to `pathlib.Path.glob`. Some benchmarking should be done however to figure out if that is really a good idea performance-wise.

foosel, 2022-06-21:
`pathlib_glob_benchmark.py`
``` python
import os
import pathlib

path = "c:/Users/Gina/Application Data/OctoPrint/uploads"

def iterate_pathlib():
    for _ in pathlib.Path(path).glob('**/*'):
        pass

def iterate_os_scandir():
    for _ in os.scandir(path):
        pass

def print_timings(raw_timings, number, repeat):
    timings = [t / number for t in raw_timings]
    best = min(timings)
    print(f"{number} iterations, best of {repeat}: {best:.10f}s")

if __name__ == "__main__":
    import timeit

    number = 10000
    repeat = 5

    print("pathlib:")
    print_timings(timeit.repeat("iterate_pathlib()", setup="from __main__ import iterate_pathlib, path", number=number, repeat=repeat), number, repeat)
    print("os.scandir:")
    print_timings(timeit.repeat("iterate_os_scandir()", setup="from __main__ import iterate_os_scandir, path", number=number, repeat=repeat), number, repeat)

```

Running this yields this result:

``` shell
$ python pathlib_glob_benchmark.py
pathlib:
10000 iterations, best of 5: 0.0001418229s
os.scandir:
10000 iterations, best of 5: 0.0000403221s
```

So, from a performance perspective, this replacement does not make sense as `os.scandir` is significantly faster. And in fact - after some digging through CPython history - is also [used by pathlib internally](https://github.com/python/cpython/blob/main/Lib/pathlib.py#L295).

Closing.

closed 2022-06-21 by foosel

frenck, 2022-06-21:
Pathlib is slower, but in general not enough to be a concern. IMHO the syntax is a huge improvement the OOP style approach will generate much more readable code and is way better validatable/lintable with things like mypy (as the results are objects).