OctoPrint/OctoPrint #4549: [Request] Bundle the System Command Editor plugin (or reimplement logic)

foosel, 2022-06-21:
### Is your feature request related to a problem? Please describe.

The System Command Editor plugin offers a UI based way to edit the commands listed in OctoPrint's system menu. It's been created and maintained by @Salandora but they no longer seem to have time for it and based on the big Warning in the [project's README](https://github.com/Salandora/OctoPrint-SystemCommandEditor) I believe it can be considered abandoned now. It is in need of maintenance.

### Describe the solution you'd like

Bundle the plugin into OctoPrint itself, but on an external release cycle to allow for easier maintenance support by the community.

### Describe alternatives you've considered

  * Losing the functionality. Not a good idea given its popularity.
  * Building an alternative right into OctoPrint. Not a good idea given the work has already been done.
  * Putting it straight into OctoPrint's source tree. Not a good idea since that would force it on the same release cycle.
  * Finding a new maintainer for it and keeping it unbundled. Would certainly be an option but I'm leaning towards this functionality meriting inclusion into the core package, given that it's repeatedly been confused with core functionality. 

Thoughts?

bkhofmann, 2022-09-29:
Sounds like a good idea.  I recently came across this plugin and am disappointed that I can't get it to work after installing.  An entry doesn't come up for it in the Plugins section under OctoPrint settings even though the plugin manager seems to think it's installed.

b-morgan, 2023-08-09:
When I installed this plugin, it came from [https://github.com/synman/OctoPrint-SystemCommandEditor](https://github.com/synman/OctoPrint-SystemCommandEditor) which appears to be a fork of the [Salandora](https://github.com/Salandora) repository.

In any case, I vote for bundling the plugin.