OctoPrint/OctoPrint #4551: Settings Object Is Not Thread Safe - And Thus Sets/Saves Can Lose Data

QuinnDamerell, 2022-06-23:
### The problem

I have had a few users reporting issues where OctoPrint will reboot into safe mode sometimes when OctoEverywhere is installed. I have been debugging it for a while, and I think I found the cause.

When OctoPrint boots into safe mode, it's because of a "failed startup". The flag that controls this is the `incompleteStartup` flag. The flag is set when the server first boots and is cleared just after `on_after_startup` is called on the plugins.

What was happening in my case was two issues. I have logic that starts a **thread** to make the main connection to my servers after `on_after_startup` is called. That thread does some work, some of which writes settings into the plugin settings object. The first problem was that I was using the `force=True` argument for all of my set and save calls, which I shouldn't have been.

But the second problem I debugged was that it seems the settings object set/save functions don't thread safe. So if it just so happened that OE was setting and saving the config the same time the system logic was saving the removal of the `incompleteStartup`, sometimes the `incompleteStartup` removal would get overwritten.

With about 15 plugins loaded on a rasp pi W2 I was about to get the race above to repo about 1 out of 10 times OctoPrint started. 

I looked into the Settings object a little, and while I don't understand it entirely, I do think the `self._dirty` flag could get into bad states since it's not protected by a lock at all. But that would be alright, the next setting write/save would correct it. I think the bigger problem is the save/set functions don't seem to be thread-safe between each other, which I think results in the data loss.

I think it would be easy to repo this more often if you wrote a test plugin that spins in a loop after `on_after_startup` that users set/save with force=True.

While I'm going to remove the force=ture flags from my logic and try to make sure every setting possible is set and written before on_after_startup returns, I think there's still some chance this could happen to OE or any plugin in the future since the settings object doesn't seem to be entirely thread-safe.

### Did the issue persist even in safe mode?

No, it did not persist (it's an issue with one of your installed plugins, don't report here)

### If you could not test in safe mode, please state why

3rd party plugins aren't enabled

### Version of OctoPrint

1.8.2

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Raspi W2 (and lots of other devices from my community)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

QuinnDamerell, 2022-06-23:
@foosel exactly! I was thinking that a lock around the set function at least, and maybe around the _dirty flag and save() function might do the trick. But like you said, it's quite a complicated object and a very important one, so I didn't want to throw something in there without understanding the entire thing well.

I'm on version of OctoPrint 1.8.2 because I have a time machine... or because I wrote this up at 11pm and my brain was tired.... I can't remember. 😝

foosel, 2022-06-28:
I've thrown up a PR for that at #4560. Would love some reviews.

foosel, 2022-08-18:
The PR has been merged and this should hopefully be a thing of the past come 1.9.0.

commit 7020a21c (2022-08-03): 🐛 Make settings threadsafe

setter, save and dirty flag access are now behind
a reentrant lock.

Fixes #4551
commit d3517944 (2022-08-03): 🐛 Make settings threadsafe (#4560)

setter, save and dirty flag access are now behind
a reentrant lock.

Fixes #4551
closed 2023-05-23 by foosel