OctoPrint/OctoPrint #4552: [Bug] pyproject.toml in a plugin depending on `octoprint_setuptools` renders it uninstallable

foosel, 2022-06-23:
In a plugin (e.g. [OctoPrint-FixCBDFirmware](https://github.com/OctoPrint/OctoPrint-FixCBDFirmware)) move `black.toml` to `pyproject.toml` and adjust the pre-commit config so the `black` run no longer looks for `black.toml`. It will take its config from `pyproject.toml`.

Then create a new venv, install OctoPrint into it and attempt to install the plugin via `pip install -e .[develop]`. It will fail because it cannot find `octoprint_setuptools`:

```shell
$ pip install -e .[develop]
Obtaining file:///D:/Code/OctoPrint/OctoPrint-FixCBDFirmware
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 4294967295:
   command: 'd:\code\octoprint\experiments\fixcbd_env\scripts\python.exe' 'd:\code\octoprint\experiments\fixcbd_env\lib\site-packages\pip\_vendor\pep517\_in_process.py' get_requires_for_build_wheel 'C:\Users\gina\AppData\Local\Temp\tmpto28quzy'
       cwd: D:\Code\OctoPrint\OctoPrint-FixCBDFirmware
  Complete output (1 lines):
  Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
  ----------------------------------------
ERROR: Command errored out with exit status 4294967295: 'd:\code\octoprint\experiments\fixcbd_env\scripts\python.exe' 'd:\code\octoprint\experiments\fixcbd_env\lib\site-packages\pip\_vendor\pep517\_in_process.py' get_requires_for_build_wheel 'C:\Users\gina\AppData\Local\Temp\tmpto28quzy' Check the logs for full command output.
```

Reverting the change makes the plugin installable again.

See also some failed builds due to that:

- https://github.com/OctoPrint/OctoPrint-FixCBDFirmware/actions/runs/2548647620
- https://github.com/OctoPrint/OctoPrint-FixCBDFirmware/actions/runs/2548753208

Something about the presence of the `pyproject.toml` file makes the execution of `setup.py` behave differently. This needs investigation since it is a very obscure error that could trip up plugin developers, plus it prevents us from switching the plugin template over to more modern tooling as well.

Admittedly I know way too little yet about the internals of said more modern tooling to have a clue about what is going on here, so it needs some research.

Since this relates to `octoprint_setuptools`, I'm opening this as a ticket in the OctoPrint core repo, though I was also briefly thinking about the new meta repo.

cp2004, 2022-06-23:
I was trying to create a python package (not plugin) that used [python-versioneer](https://github.com/python-versioneer/python-versioneer), but as soon as I created `pyproject.toml` it failed to install because it couldn't import versioneer anymore. Similar kind of problem.

It seems that when pip encounters a pyproject.toml, it wants to build it in isolation and then installs it later.

foosel, 2022-06-23:
Ah, good to know that's a general thing then, and that just have me the right words to Google and I came across this:

https://pip.pypa.io/en/stable/reference/build-system/pyproject-toml/

So, yep, builds are done in isolation in this case, so I guess the solution would be to define OctoPrint as a build time dependency.

I do far didn't make it one because I wanted to prevent people from trying to install an OctoPrint plugin without having OctoPrint *already* (in order to discover things like manual pip install attempts without first activating the venv), but maybe there is a different way too solve this, or maybe the answer is just to extract `octoprint_setuptools` into its own package on pypi.

foosel, 2022-11-10:
So, moving `octoprint_setuptools` into its own pypi package sadly doesn't solve this for `setup.py` based isolated installs. `setup_requires` in `setup.py` can (obviously) only be evaluated once `setup` has been called. So when (like in plugins and OctoPrint itself) the `octoprint_setuptools` stuff is required at build time, for fiddling together the parameters to `setup`, stuff won't work.

Switching plugins over to `pyproject.toml` might be the clean solution here. Will require a TON of testing though across several Python environments, pip versions, setuptools etc. What a nightmare this build isolation is going to be 🤦‍♀️

foosel, 2022-11-14:
I've extracted `octoprint_setuptools`. So plugins that already use or want to switch to using `pyproject.toml` can now do this by adding this to `pyproject.toml`:

    [build-system]
    requires = ["setuptools>=40.8.0", "wheel", "octoprint-setuptools"]
    build-backend = "setuptools.build_meta"

This makes sure that the `octoprint-setuptools` dependency will be installed into the isolated build environment that gets triggered by the presence of `pyproject.toml`. `setup.py` based installs in a non-isolated environment will still be able to just use `octoprint_setuptools` as pulled in by OctoPrint itself. And a new dependency on `wheel` will also make sure that no isolation should happen for building the wheel of `setup.py` based plugins anymore, see #4682.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel