OctoPrint/OctoPrint #4554: File that makes the gcode analyzer crash

cp2004, 2022-06-24:
### The problem

I have this file in my testing OctoPrint install that someone sent to do some debugging (I think of a separate issue):
[np.aw (1).zip](https://github.com/OctoPrint/OctoPrint/files/8981731/np.aw.1.zip)

It produces the following result:

```
Traceback (most recent call last):
  [....]
  File "c:\users\charl\development\octoprint core\octoprint\src\octoprint\cli\analysis.py", line 123, in gcode_command
    interpreter.load(
  File "c:\users\charl\development\octoprint core\octoprint\src\octoprint\util\gcodeInterpreter.py", line 300, in load
    self._load(
  File "c:\users\charl\development\octoprint core\octoprint\src\octoprint\util\gcodeInterpreter.py", line 681, in _load
    fwretractTime = s / f
ZeroDivisionError: float division by zero
2022-06-24 21:24:42,486 - octoprint.filemanager.analysis - ERROR - Analysis for local:np.aw.gcode ran into error: No analysis result found
```

Maybe a bug in the firmware retract handling (if that is present in the file).

I haven't investigated the file yet to see what may be the cause, but @jneilliii has confirmed the issue is not something I messed up locally...

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

Currently a 1.9.0dev version, testing the new webcam plugin interface

### Operating system running OctoPrint

Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

My systeminfo bundle is going be full of absolute crap because of the development, but here it is to not upset the bot, but the full error is in there 

[octoprint-systeminfo-20220624231708.zip](https://github.com/OctoPrint/OctoPrint/files/8981744/octoprint-systeminfo-20220624231708.zip)

cp2004, 2022-06-25:
At the end of the file there is this line:

```
M207 S0 Z0 F0 ; reset FW retractions
M208 S0 F0 ; reset FW recover
```

Causing the problem

foosel, 2022-06-27:
Fixed by #4555 by @cp2004, will be part of 1.9.0.

PR #4555 (2022-06-27): :bug: gcodeInterpreter: Avoid ZeroDivisionError with firmware retraction

#### What does this PR do and why is it necessary?

Avoid crashing if the F parameter is 0, through a ZeroDivisionError

#### How was it tested? How can it be tested by the reviewer?

See file in issue #4554 for reproduction and now confirm it is fixed

#### What are the relevant tickets if any?

Closes #4554
commit 151c38a5 (2022-06-27): :bug: gcodeInterpreter: Avoid ZeroDivisionError with firmware retraction (#4555)

Closes #4554
closed 2023-05-23 by foosel