OctoPrint/OctoPrint #4562: [Request] Drop ES5 restriction on core UI and switch to ES6 compatibility and thus allow modern JS language features

foosel, 2022-06-29:
### Is your feature request related to a problem? Please describe.

OctoPrint has been limited to ES5 functionality for years now. In #3920 I even added linting and polyfills to ensure the core code would stay ES5 compatible. However, as rightly pointed out in #4559, ES6 has been a thing now for ages, and based on the stats of the [Anoymous Usage Tracking](https://tracking.octoprint.org) as of 2022-06-29, 99.1% of all OctoPrint clients already support ES6 (source: [comment on #4559](https://github.com/OctoPrint/OctoPrint/pull/4559#issuecomment-1169216486)):

> Alright, that are some useful statistics.
> |Type|number|percentage|
> |:----|----:|----:|
> |Total Instances:  |   175974 ||
> |ES6 Support:             | 174438 | 99.13% |
> |No Support:        |     399  | 0.23% |
> |Unknown support:  |     1137 | 0.65% |
> 
> This is the script that calculated the statistics. If you can identify which versions of WebKit support ES6 you can get 424 extra instances classified. [convertstatistics.js.txt](https://github.com/OctoPrint/OctoPrint/files/9004658/convertstatistics.js.txt)

Limiting the core UI code to ES5 levels prevents us from using things like async/await, let, const, arrow functions etc, all of which make maintenance easier and very much improve the quality of life of the maintainers as well.

### Describe the solution you'd like

From a total of 176k client connections, only 400 or 0.23% didn't support ES6 features. Adding in unknown support, we arrive at 0.88% unknown or non existing ES6 support. I think at this point it is fair to say, it's time those 0.23-0.88% to switch to a modern browser instead of letting OctoPrint as a whole stay stuck in ES5 times for everyone.

So at the very least the restrictions on OctoPrint's core UI code to stick to ES5 should be lifted. ESLint should be set to ES6 compatibility mode and in forthcoming development modern language features should be used.

Additionally existing code could also be migrated, however, since long term we want to get rid of the existing core UI anyhow and replace it with something react based, I'm not sure the immense refactoring effort caused by this would be worth it.

If we go for this, what we definitely should make available/link to is some page for OctoPrint's users to figure out if their browser supports ES6 (and possibly other stuff in the future). caniuse.com has the data, and https://caniuse.com/es6 holds the info on ES6 specifically, but I couldn't yet locate a "you are here" kinda feature. Any hints in that regard would be highly appreciated.

### Describe alternatives you've considered

Staying on ES5 for all eternity or at least until a new frontend gets added. Kinda meh given that this still hasn't gotten beyond the brainstorming phase and some testing, and there are also still some big issues to figure out (plugin support), so making a cut now to make maintenance and development less of a pain for everyone involved - just like with dropping Python 2 support - would seem a wise choice.

### Additional context

See the discussion that sparked this request in #4559. Also see the stats exports at https://data.octoprint.org/export/ that are the basis for the above statistical evaluations.

JoveToo, 2022-06-29:
I've made a very basic page that test for:
* let
* async/await
* generator functions
It should weed out none es6 browsers pretty quickly.

[es6quicktest.html.txt](https://github.com/OctoPrint/OctoPrint/files/9014200/es6quicktest.html.txt)

foosel, 2022-08-02:
ESLint config has been adjusted to allow ES6 code. Two files have been ported to use arrow functions as a test.

Still todo: either a static page somewhere on the octoprint.org domain, or alternatively something built into OctoPrint, that tests ES6 compatibility. Currently leaning to a simple static page on octoprint.org, very much like the one shared by @JoveToo above.

foosel, 2023-02-22:
Almost forgotten, but thankfully still remembered while working on the 1.9.0rc1 changelog. Test page is now up at https://octoprint.org/es6test/

foosel, 2023-03-06:
> I'm kinda surprised it didn't raise concerns about the use of `async/await` under ES6 constrictions

Well, I can answer that for myself: the constraints were EcmaScript 2018 instead of 2015. Setting it to EcmaScript 2017 however now raises an error due to the use of `async function *` and indeed, async generator functions were only introduced in ES2018. So, we are sticking to it but data.octoprint.org, the changelog and the browser check need to be adjusted accordingly.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel