OctoPrint/OctoPrint #4566: HLS Stream not displayed in dashboard, but works fine in stream test dialog

cryptoAlgorithm, 2022-07-05:
### The problem

1. Enabled HLS webcam streaming (after hours of debugging 64-bit ffmpeg issues)
2. Change the stream URL in settings to reflect the new URL
3. Verify the stream works by clicking on 'Test' (see screenshot below)
4. Go back to the dashboard, both control tab and webcam item in sidebar does not work

I have reloaded and restarted OctoPrint multiple times to no avail. However, I noticed requests are being made to the stream in DevTools, but the stream never loads.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

OctoPi v1.0.0

### Printer model & used firmware incl. version

None (virtual printer)

### Browser and version of browser, operating system running browser

Chrome 102.0.5005.61, same issue on Safari 16.0 (on macOS 13.0)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

#### Sidebar stream not visible in Control tab
[image]

#### Stream never loads in sidebar
[image]

#### Requests being made to load stream in DevTools
[image]

#### Stream works in test dialog
[image] 

## Systeminfo
[octoprint-systeminfo-20220705130351.zip](https://github.com/OctoPrint/OctoPrint/files/9043388/octoprint-systeminfo-20220705130351.zip)

## Console logs
```
Uncaught (in promise) Objectmessage: "A listener indicated an asynchronous response by returning true, but the message channel closed before a response was received"[[Prototype]]: Object
7[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
packed_core.js?dab326b1:1304 Starting dependency resolution...
packed_core.js?dab326b1:1317 ... dependency resolution done
packed_core.js?dab326b1:1340 Initial application setup done, connecting to server...
packed_core.js?dab326b1:1135 Connected to the server
packed_core.js?dab326b1:1140 Triggering connect on all view models
packed_core.js?dab326b1:1142 Server (re)connect processed
packed_core.js?dab326b1:1340 Finalizing application startup
packed_core.js?dab326b1:1321 Going to bind 44 view models...
packed_core.js?dab326b1:1333 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?dab326b1:806 Chrome Arc Fix disabled
packed_core.js?dab326b1:806 Chrome Arc Fix enabled
packed_core.js?dab326b1:1333 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?dab326b1:1333 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
[... 64 lines cut ...]
callViewModelIf @ packed_core.js?dab326b1:1268
DevTools failed to load source map: Could not load content for chrome-extension://hnmpcagpplmpfojmgmnngilcnanddlhb/browser-polyfill.min.js.map: System error: net::ERR_BLOCKED_BY_CLIENT
DevTools failed to load source map: Could not load content for http://octopi.local/plugin/uicustomizer/static/js/chartist.min.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
packed_core.js?dab326b1:1268 Error calling onBrowserTabVisibilityChange on view model ControlViewModel : TypeError: Cannot set properties of null (setting 'onresize')
    at ControlViewModel.self._switchToHlsWebcam (http://octopi.local/static/webassets/packed_core.js?dab326b1:121:255)
    at ControlViewModel.self._enableWebcam (http://octopi.local/static/webassets/packed_core.js?dab326b1:115:133)
    at ControlViewModel.self.onBrowserTabVisibilityChange (http://octopi.local/static/webassets/packed_core.js?dab326b1:115:773)
    at OctoPrint.coreui.viewmodels.controlViewModel.onBrowserTabVisibilityChange (http://octopi.local/static/webassets/packed_plugins.js?ca2abbec:14565:37)
    at callViewModelIf (http://octopi.local/static/webassets/packed_core.js?dab326b1:1267:75)
    at http://octopi.local/static/webassets/packed_core.js?dab326b1:1261:138
    at Pn (http://octopi.local/static/webassets/packed_libs.js?7111420c:30:530)
    at Function.<anonymous> (http://octopi.local/static/webassets/packed_libs.js?7111420c:51:66)
    at callViewModelsIf (http://octopi.local/static/webassets/packed_core.js?dab326b1:1261:95)
    at callViewModels (http://octopi.local/static/webassets/packed_core.js?dab326b1:1260:56)
callViewModelIf @ packed_core.js?dab326b1:1268
(anonymous) @ packed_core.js?dab326b1:1261
Pn @ packed_libs.js?7111420c:30
(anonymous) @ packed_libs.js?7111420c:51
callViewModelsIf @ packed_core.js?dab326b1:1261
callViewModels @ packed_core.js?dab326b1:1260
(anonymous) @ packed_core.js?dab326b1:1338
(anonymous) @ packed_core.js?dab326b1:1290
Pn @ packed_libs.js?7111420c:30
(anonymous) @ packed_libs.js?7111420c:51
updateBrowserVisibility @ packed_core.js?dab326b1:1290```

cryptoAlgorithm, 2022-07-05:
Update: sidebar stream works right after reloading, but not in control tab. Switching to the control tab then back to another tab leads to the continuous loading issue.

[image]

cp2004, 2022-07-05:
The sidebar stream is provided by a plugin, not core OctoPrint. When you are running in safe mode all the plugins you have should be disabled and you are back to the default UI. None of the screenshots here are showing what it is looking like with the standard UI. Can you test in safe mode, and provide the screenshots when all the plugins are disabled?

jneilliii, 2022-07-05:
Yeah, I suspect this to be an issue with UI Customizer.

cryptoAlgorithm, 2022-07-06:
@cp2004 the screenshots provided in the OP are with safe mode activated, as evident from the warning at the top left. The follow up screenshot was without safe mode, but the issues persisted with or without safe mode.

cp2004, 2022-07-06:
> In case this makes any difference, I do notice the webcam in the sidebar is present from the get-go right out of the box, before installing any additional plugins. Could this be something octopi-related?

That is *very* strange behaviour there - the sidebar webcam is *definitely* provided by UI Customizer in your case, and that doesn't come with the stock OctoPi image. I will admit to being rather confused by the problem you have here, I had not imagined it would be possible!

A stock setup would have the webcam just on the control tab, and no right-side sidebar, a little like this:
[image]

cryptoAlgorithm, 2022-07-07:
@cp2004 unfortunately I left the octopi setup at a remote location, so can't verify it again. But I am certain safe mode disabled UI customizer, since it no longer says UI customizer in the loading screen. The dashboard also does not have any of the other improvements that plugin brings like the different styling for the bottom bar or mobile layout optimisations.

closed 2022-07-07 by jneilliii

cryptoAlgorithm, 2022-07-08:
@jneilliii got it. I just installed OctoPrint on my local machine, and it doesn't have the sidebar. Maybe something that comes preinstalled in octopi is causing that?

cp2004, 2022-07-08:
There's no extra plugins installed with the official OctoPi image, which you can find on https://octoprint.org/download or in the Raspberry Pi imager.