OctoPrint/OctoPrint #4578: depth: 1.0e+18 makes web interface not load and browser unhappy

jaymemaurice, 2022-07-14:
### The problem

1. Install fresh virgin docker octoprint container or whatever
2. Step through everything until you get to printer dimensions
3. Pretend you have a belt printer and put 9999999999999999 in the Y axis (exact number of 9's needed unknown...)
4. Finish the wizard
5. note that the web interface is frozen. navigating back to octoprint will just make browser go unresponsive. no ability to restart in safe mode from browser.
6. note that _default.profile has an exponent in depth eg docker exec -it octoprint cat /octoprint/octoprint/printerProfiles/_defualt.profile - not sure if a big number represented without exponent will work or not, did not test.
7. fix the config file and restart; eg. docker exec -it octoprint /bin/sed -i "s/1.0e+18/900/" /octoprint/octoprint/printerProfiles/_defualt.profile; docker restart octoprint
8. Note issue does not occur with reasonable dimension

Suggest wizard enforce "sane?" printer dimensions and any process which attempts to use the build volume force sane limit

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

didn't install any plugins

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

debian

### Printer model & used firmware incl. version

NA/CR-30

### Browser and version of browser, operating system running browser

Chrome/Chromium

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Javascript console showed nothing interesting...
Profiler shows browser stuck in drawRectangularGrid
[image]

jneilliii, 2022-07-14:
Curious if you disable the gcode viewer plugin in Plugin Manager does the same issue re-occur? I could see this being something related to the rendering process based on the function call.

jaymemaurice, 2022-07-14:
I'll give it a shot today.

foosel, 2022-11-03:
Was indeed caused by the viewer attempting to render billions of grid lines. I've limited the value range for width, depth and height to 9999mm. If someone shows up here who has a printer with a build volume of a km side length, we can always talk about making this configurable :P

Fix ready for 1.9.0.

closed 2023-05-23 by foosel