OctoPrint/OctoPrint #4582: G02 or G03 are not interpreted as arcs in the gcodeviewer

cp2004, 2022-07-18:
### The problem

G2 and G3 commands are correctly rendered as arcs, however G02 and G03 commands are not and just render as direct straight lines. Discovered while testing #g02

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

#4580

### Operating system running OctoPrint

Windows 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220718155606.zip](https://github.com/OctoPrint/OctoPrint/files/9132967/octoprint-systeminfo-20220718155606.zip)

kForth, 2022-07-19:
This seems to be a result of the following lines:

https://github.com/OctoPrint/OctoPrint/blob/a2e9aefe9c8a51120931e9b14bb95ebf9ea6c7f5/src/octoprint/plugins/gcodeviewer/static/js/viewer/worker.js#L482-L485

https://github.com/OctoPrint/OctoPrint/blob/a2e9aefe9c8a51120931e9b14bb95ebf9ea6c7f5/src/octoprint/plugins/gcodeviewer/static/js/viewer/renderer.js#L1074

The `.charAt(1)` in both checks will return `0` for `G02` and `G03` commands, meaning `direction` doesn't get set here and will be `undefined` when it's time to render.

I've created PR #4587 to fix this.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel