OctoPrint/OctoPrint #4583: Web portal tab element hover style moves the tab slightly, sometimes causing oscillation.

msandy, 2022-07-18:
### The problem

1. Navigate to the main page of the web portal.
2. Hover the cursor over the bottom edge of one of the tab buttons.

Expected: tab highlight style is temporally stable

Actual: tab highlight style oscillates with some cursor positions

This is almost certainly due to the hover-style making the dimensions or margins of the object smaller than the non-hover style, causing the hover state to oscillate.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.8.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

N/A - web interface issue

### Browser and version of browser, operating system running browser

N/A - repros on both Chrome on Mac and Edge on Windows

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

https://user-images.githubusercontent.com/12552462/179560463-80f8fc8f-d4cc-4dc0-a22e-d77fd453a220.mp4

https://user-images.githubusercontent.com/12552462/179560466-047541bc-a037-403a-8a1c-f97cc9da7cf9.mo

[js.log](https://github.com/OctoPrint/OctoPrint/files/9133771/js.log)

[octoprint-systeminfo-20220718094109.zip](https://github.com/OctoPrint/OctoPrint/files/9133737/octoprint-systeminfo-20220718094109.zip)

cp2004, 2022-07-18:
Did you actually test this in safe mode?

I can't reproduce this, and your reproduction videos are not recorded in safe mode.

msandy, 2022-07-18:
oops - looks like I was a little impatient and didn't wait for the reboot. It looks like this is actually caused by Themeify.

closed 2022-07-18 by msandy

jneilliii, 2022-07-18:
recommend switching to actively developed theme plugin UI Customizer.