OctoPrint/OctoPrint #4585: Controls configuration being routinely wiped out

AapoTahkola, 2022-07-18:
### The problem

The custom control buttons defined in config as:
```
controls:
- children:
    - children:
      - commands:
        - M300 S100 P300 ; beep
        - M290 Z0.25
        confirm: null
        name: 0.25 baby up
```
Are routinely removed by "someone" presumably by octoprint. This has happened three times already and I am getting tired of having to rewrite them.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

no other plugins enabled, it should work in normal mode

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

raspbian

### Printer model & used firmware incl. version

Marlin bugfix-2.1.x

### Browser and version of browser, operating system running browser

firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220718233853.zip](https://github.com/OctoPrint/OctoPrint/files/9135618/octoprint-systeminfo-20220718233853.zip)

AapoTahkola, 2022-07-18:
As a workaround I did "chmod -r /home/pi/.octoprint/config.yaml". Not sure if it is going to do anything.It seems to at least couple of days before the bug gets triggered.

jneilliii, 2022-07-18:
Your issue is a bug with custom controls plugin. Once you get your controls configured the way you like disable the plugin and it won't wipe out your settings. Or alternatively, install my patched version by copying/pasting the URL below into plugin manager > get more > ...from URL and click install.

```
https://github.com/jneilliii/octoprint-customControl/archive/refs/heads/devel.zip
```

AapoTahkola, 2022-07-18:
Why is not Custom Control Editor plugin marked as abandoned? I did indeed disable it once already.

> Your issue is a bug with custom controls plugin. Once you get your controls configured the way you like disable the plugin and it won't wipe out your settings. Or alternatively, install my patched [...]

Thanks I will use this instead.

jneilliii, 2022-07-18:
We've decided to create a settings interface for custom controls in a future release of OctoPrint, but no one has started working on the integration yet. I believe there is an issue on the plugin repo about marking the plugin abandoned, just need to flag it as such as it wasn't until recently the plugin author replied back to us.

AapoTahkola, 2022-09-22:
After using this plugin for a while I have stopped using all of the nested horizontal, vertical stuff and just do a straight out list. I could suggest that the new custom control editor is stripped out of all these functionality and left with just option to move items up and down in the list. It just adds complexity while delivering very little value to the end user.
 I can easily find what I am looking for in 20 item list and if it were supported I see no problem even 40 or 60 custom controls.