OctoPrint/OctoPrint #4586: Include Marlin capabilities or full M115 output in the logs

PowerWiesel, 2022-07-18:
### Is your feature request related to a problem? Please describe.

It would help us to find out if certain features are enabled in Marlin without asking the user to enable serial logging or to copy that part from the terminal.

### Describe the solution you'd like

Include Marlin capabilities or full M115 output in the logs.

foosel, 2023-07-03:
Implemented by the first of the two above commit. The second also adds a new file `m115.txt` to the system info bundle (if said bundle is being generated while there is an active printer connection and firmware data was received), which will contain a reconstructed version of the M115 response received from the firmware, based on the already parsed data.

E.g.:

```
Reconstructed M115 response:

>>> M115
<<< FIRMWARE_NAME:Virtual Marlin 1.0 PROTOCOL_VERSION:1.0
<<< Cap:AUTOREPORT_TEMP:1
<<< Cap:AUTOREPORT_SD_STATUS:1
<<< Cap:AUTOREPORT_POS:0
<<< Cap:EMERGENCY_PARSER:1
<<< Cap:EXTENDED_M20:1
<<< Cap:LFN_WRITE:1
```

`octoprint.log` also now contains the data:

```
2023-07-03 17:19:50,469 - octoprint.util.comm - INFO - Printer reports firmware name "Virtual Marlin 1.0"
2023-07-03 17:19:50,470 - octoprint.util.comm - INFO - Firmware info line: FIRMWARE_NAME:Virtual Marlin 1.0 PROTOCOL_VERSION:1.0
2
[...]
2023-07-03 17:19:50,541 - octoprint.util.comm - INFO - Firmware sent the following capability report:
  AUTOREPORT_TEMP: supported
  AUTOREPORT_SD_STATUS: supported
  AUTOREPORT_POS: not supported
  EMERGENCY_PARSER: supported
  EXTENDED_M20: supported
  LFN_WRITE: supported
```

Will be shipped with 1.10.0

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel