OctoPrint/OctoPrint #4589: Got a resend request from the printer

Mech131, 2022-07-19:
### The problem

1.  When the printer (Ender 7) receives a resend request, the tool moves outside of the bounds of the print space and the motors start to grind.
2. Issue does not occur when printing from SD Card.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender7_HW-2.4.S1.101_SW-1.0.4-noBLTouch_220308

### Browser and version of browser, operating system running browser

Edge / Win 11

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220719161903.zip](https://github.com/OctoPrint/OctoPrint/files/9144823/octoprint-systeminfo-20220719161903.zip)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/9144824/octoprint.log)
[Uploading Pot 61 Water Bowl.zip…]()

cp2004, 2022-07-20:
> Did the issue persist even in safe mode?
Yes, it did persist

Forgive me if you did test this in safe mode and it is just not in there, but the systeminfo bundle is not reporting any safe mode tests there, so can I ask again if you *actually* tested it in safe mode?

---

Anyway, on to what the actual problem is. I looked through the logs, and every time a resend request occurred it was because the line was over 64 characters/bytes in length and the printer had an issue reading that, the start of the command was cut off. When you are using the 'Arc Welder' plugin, this can increase the line lengths because it has more parameters to fit in. See the examples pulled out the logs below.

Since you are using Creality's provided firmware, all I can recommend is to run in the opposite direction, fast. They seem to have set the max line length to 64 on your printer. Every time an issue like this comes up it seems to be Creality, build your own Marlin or use a pre-built one such as this one from [Insanity Automation](https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/Hex%20Files/firmware_Ender7_DW7.4.6.bin.zip). 

```
| Send: N572020 G2 X80.509 Y144.879 I245949.690 J3824.193 E1771.51005 F3600*63
(64 characters here completely cuts off the line number, so it is not an invalid command just the number is gone)

| Send: N144967 G3 X172.879 Y138.555 I-0.086 J0.365 E1990.31496 F15000*30
| Recv: echo:Unknown command: "4967 G3 X172.879 Y138.555 I-0.086 J0.365 E1990.31496 F15000"

| Send: N161169 G2 X89.362 Y146.754 I0.024 J0.029 E2825.61421 F15000*11
| Recv: echo:Unknown command: "161169 G2 X89.362 Y146.754 I0.024 J0.029 E2825.61421 F15000"

| Send: N224693 G3 X80.061 Y145.312 I-11681.706 J-160.648 E1870.28457 F4800*63
```

Also:

You have an 'M73 Progress' plugin sending commands to the printer, but the printer doesn't seem to support that.
```
| Send: N224690 M73 P35*90
| Recv: echo:Unknown command: "M73 P35"
```

**Next time please use the forums for support with things like this**. This issue tracker is for bugs & feature requests for OctoPrint. For help and support in using OctoPrint, plugins, your printer and your general setup please get in touch on the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org) - helpful people are around that can answer your questions.

closed 2022-07-20 by cp2004

Mech131, 2022-07-20:
@cp2004 Forgive me for not quite understanding how to solve this issue. It would appear that I need to disable arc-welder and reupload my g-code files? The other solution is to build my own Marlin? I tired using IA's firmware, but there are several bugs that make it unusable on the Ender 7. I've reached out over discord to the firmware team and they are aware of the bugs, but don't have an ETA on the fix. 

This probably isn't the right place to ask, but for the sake of saving time; any idea where I would modify the readable line lengths in Marlin?

cp2004, 2022-07-20:
I think you understand it alright. The default max size mainline Marlin is 96, I can't see what it is in Creality's firmware that they've changed but the consistency across the 4 resend issues makes me confident it looks like 64. That's in the file linked here:
https://github.com/MarlinFirmware/Marlin/blob/aacc2d3dc5657b17a26cde09d0146486ea72b17e/Marlin/Configuration_adv.h#L2320

The alternative with Arc Welder - I did just remember that AW actually has a setting for this, so either disabling the plugin or try setting 'Maximum Gcode Length' to something like 50. That should account for the line numbers & checksums added afterwards:
[image]