OctoPrint/OctoPrint #4592: CORS issue with .local address in chrome windows

KilianB, 2022-07-22:
### The problem

Navigating to `octoprint.local` fails to load relevant scripts on chrome on my windows machine, thus I am not able to log in.
With apple it's working fine.

**Error Log:** 

````
Access to script at 'http://octopi.local/static/js/lib/bootstrap/bootstrap.js' from origin 'http://octopi.local' has been blocked by CORS policy: The request client is not a secure context and the resource is in more-private address space `private`.
````

[octoprint-systeminfo-20220722215400.zip](https://github.com/OctoPrint/OctoPrint/files/9172083/octoprint-systeminfo-20220722215400.zip)

**Workaround:**

The page can perfectly fine be accessed via the ip address or the https:// side (with notice of invalid certificate). I am not well versed in network scopes and can't say if this is just related to a network setup issue on my side. 
Ref: https://wicg.github.io/private-network-access/#ip-address-space-less-public

Adding the header [Access-Control-Request-Private-Network: true ](https://wicg.github.io/private-network-access/#http-headerdef-access-control-request-private-network)might resolve the issue but I do not understand the ramifications of this. 

Does octoprint host it's own web sever or should this ticket be opened in the octopi repo?

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

OctoPi 0.18.0

### Browser and version of browser, operating system running browser

Win 10 : Chrome: Version 103.0.5060.114 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

KilianB, 2022-07-22:
Logging into the page once via https:// or upgrading to the newest 1.8.1 or some kind of cache issues resolved the issue on my side

closed 2022-07-22 by KilianB

cp2004, 2022-07-22:
I think it is a browser bug - I have just spent some time searching through the chromium bugs site.

Fairly obviously (to you and me at least) CORS shouldn't apply here - both requests are from the *same* origin, not *cross* origin. Not too long ago both @jneilliii & I had similar issues with the resource address space being identified as 'unknown' randomly (for me like 1 in 100 times loading the page). Many issues should be fixed with Chrome 103 & 104. I guess there may still a bug somewhere, there was a lot of tiny problems they had with this from the reading I've just done.

Please always update to the latest version of OctoPrint before reporting bugs - we may have fixed it already 🙂