OctoPrint/OctoPrint #4594: G-Code Viewer doesn't show a lot of extrusions

gudvinr, 2022-07-23:
### The problem

1. Load print file
2. Open G-Code Viewer

A lot of moves aren't visible at all. At first I thought that it is related to #4582 but it happens with and without arcs in gcode.
Screenshot of first layer at the bottom.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

Debian 10

### Printer model & used firmware incl. version

Artillery Genius Pro, Marlin 2.1

### Browser and version of browser, operating system running browser

Firefox 102.0.1 (64-bit), Arch Linux

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Screenshot from slicer:
[image]

Screenshot from g-code viewer:
[image]

Systeminfo Bundle:
[octoprint-systeminfo-20220723230450.zip](https://github.com/OctoPrint/OctoPrint/files/9174553/octoprint-systeminfo-20220723200450.zip)

G-Code:
[phoneholderv2.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/9174554/phoneholderv2.gcode.txt)

JavaScript console:
[console-export-2022-7-23_20-9-38.txt](https://github.com/OctoPrint/OctoPrint/files/9174565/console-export-2022-7-23_20-9-38.txt)

cp2004, 2022-08-02:
Had a chance to download & test the gcode provided today, been away for a bit without access to any OctoPrint installs...

It seems to be a vase mode (spiralized) type print, where it is printing in a spiral with continuously increasing Z movements. The gcode viewer will render each of these as a separate layer, which it seems to be doing correctly. Below is a section of the gcode file, spiraling it's way up the thousands of layers:
[image]

I haven't been able to reproduce part of it that looks like your screenshot above exactly, the only thing I could get close to it was layer 2, which seems to be working fine - admittedly I didn't look at all 20,000 layers...

[image]

Can you say exactly where in the gcode file (layer/Z height?) the issue occurred? So that maybe I can analyze it closer to see potential problems

gudvinr, 2022-08-03:
@cp2004 it is very first layer that is extruded on build plate.

Also yes, this happens with any file, not just ones sliced in vase mode. I can provide test gcode for those too

cp2004, 2022-08-03:
I'm still not able to reproduce it - the file that is attached to this issue seems to different from the one you are screenshotting. I have different layer numbers for the Z height, because there is a prime line at Z1.5 first. It seems here it moves directly from the home point to start the print in your screenshot, but this is not evident in the file uploaded to the original issue.

gudvinr, 2022-08-03:
No, for latest screenshot I even downloaded file from github and uploaded to octoprint to make sure I am screenshoting same file.

cp2004, 2022-08-03:
Well, I have the same hash on the file here so that's good. All the model & layer information is different to mine, however that will be impacted by the problem at hand - it is seeming to skip parts of the file.

I've tried it on the exact same browser versions (Firefox 102) but I can't test on Arch Linux to make it *exactly* the same. Do you have another browser or device that you can test to see if we can narrow down if it is a browser issue?

gudvinr, 2022-08-03:
I also use arch linux and now issue still persist with Firefox 103.0.1 that was updated recently. Both under wayland and xwayland.

I installed latest chromium and I got same issue there too. As well as on other device that runs firefox.

[image]

I don't know how octoprint processing this gcode but since I see issues on  every browser I tested I think it might be related to octo host. I use relatively weak SBC with Allwinner A13:
```
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              1
On-line CPU(s) list: 0
Thread(s) per core:  1
Core(s) per socket:  1
Socket(s):           1
Vendor ID:           ARM
Model:               2
Model name:          Cortex-A8
Stepping:            r3p2
CPU max MHz:         1008.0000
CPU min MHz:         432.0000
BogoMIPS:            1001.47
Flags:               half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32
```

foosel, 2022-08-04:
Just to chime in, can confirm @cp2004|s observation, cannot reproduce either. But given that your model size, layer counts etc look completely different from mine, please share a screenshot from your GCode Viewer Settings page, e.g.:

[image]

I'm guessing you have something in Skip to this maybe that would explain the differences in rendering here.

gudvinr, 2022-08-04:
@foosel yep, it has `;LAYER_CHANGE` in there to skip filament purge routine.

I tried to remove it and it makes g-code viewer display layers correctly:
[image]

So I guess that skipping to first layer makes gcode viewer also ignore some stuff that it uses or Z movements during filament purge routine mess with position that viewer expecting.

foosel, 2022-08-04:
I'd rather say your first layer change comment has been added by the slicer after the actual layer change, and that is causing the issue here. I'd suggest to look for something else to match against (`; END (start_gcode)` looks promising), this is apparently not working for your particular setup and slicer. As the description of the feature says, "all GCode up to that point will be skipped". That means *absolutely* all. OctoPrint will handle the file as if it only starts with the found skipUntilThis text. If you don't want this, you'll have to live with getting a visualization of the actual file you are printing, including the purge line.

As it looks like everything is working as expected, I'm closing this.

closed 2022-08-04 by foosel

gudvinr, 2022-08-04:
> I'd rather say your first layer change comment has been added by the slicer after the actual layer change

No, if you look at gcode, it goes _before_ layer change. First it adding comment, then moves to `Z.2`
```
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G92 E0.0

G1 Z.2 F3000
```

gudvinr, 2022-08-04:
> As the description of the feature says, "all GCode up to that point will be skipped". That means absolutely all.

I am not denying that.

I tried to find what exactly breaks gcode viewer and found that if you skip to gcode after `G90` it will break.  
Which will break **all** gcode files produced by prusa slicer if you simply follow example that is also in description you are referring to:
> Example: `; layer 1`

For comparison, that is default gcode that PS inserts _before_ layer change:
```
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
```

There are no mentions that you shouldn't skip `G90` anywhere either.

foosel, 2022-08-04:
I had a look at the file, and the issue is most likely that by skipping everything up until the first LAYER_CHANGE comment, you are also skipping where it sets the extrusion to relative positioning instead of absolute. The other setup (metric units, absolute positioning on X, Y, Z) are set to default values anyhow, but the extruder being set to relative but defaulting to absolute (because that part is cut off) is then of course wreaking havoc with the extrusion detection of the rest of the file. 

The maybe good news here is that we are thinking about killing this skipping feature again anyhow as it is causing *tremendous* problems with performance and memory consumption improvements on the viewer (cc @JoveToo). 

The example is an example that does not match the output generated by PrusaSlicer, it's a generic layer comment. We don't tell the user to literally put what you put in there, we tell them that they can put stuff in there if they want to skip the whole file up to the first match. It's up to the user to decide what to put in there. I can put a reminder in there that this includes any and all prior configuration codes as well, but in the end it is the user's decision to configure this and thus manipulate what the viewer gets to visualize, and we can't hold hands on everything.

foosel, 2022-08-04:
1.9.0 will ship like this:

[image]

gudvinr, 2022-08-04:
That is good clarification, thanks. This surely won't help past me but at least it make it clear that you can break stuff.

Also issue was not because `G90` but because there is `M83` is missing. I tried gcode from simplify 3d and cura and both of them use implicit absolute positioning (there is no `G90` in produced files).
But they also use absolute extrusion mode and I suspect that octoprint also uses this by default and that is why skipping to first layer doesn't break cura gcode, for example.

foosel, 2022-08-04:
As I said here ;)

> I had a look at the file, and the issue is most likely that by skipping everything up until the first LAYER_CHANGE comment, you are also skipping where it sets the extrusion to relative positioning [...]

gudvinr, 2022-08-04:
Oh, I skimmed through that somehow.

Thank you for taking the time to deal with this. I hope that it will help other unfortunate folks to save some time in the future.