OctoPrint/OctoPrint #4595: Receiving from serial sometimes hangs for up to 5s

Florin-Popescu, 2022-07-25:
### The problem

Hello!

I have found an issue which has caused me many headaches the last few days and googling I have seen some people having similar issues but with couldn't find a solution which applies to me.

My problem is the printer pausing momentarily while printing, which produces small blobs because it's oozing and keeping the hot end in contact with my part.

I also saw that sometimes (not always) when this happens I would get a M112 sent by OctoPrint, other times it would just resume printing, depending on if OctoPrint detected a timeout or if the delay was not long enough. Changing `What to do on a firmware error` from `Disconnect from the printer` to `Cancel any ongoing prints but stay connected to the printer` caused OctoPrint to stop sending M112 so that the print would at least always resume instead of stopping dead in its tracks, but the issue itself with the printer stuttering and blobbing still was not fixed.

I have a 20min print which I can reliably use to trigger this. Always in different parts of the print, so it is not an issue with the gcode itself.

Looking at `serial.log`, I have discovered that OctoPrint would sometimes not be able to communicate anything on serial for a few seconds, and then suddenly receive all messages which should've come earlier in one quick burst. How I can easily tell is by the fact that `Not SD printing` should be sent by the printer every 1s, and indeed it can be seen that way in the logs. However when the issue occurs, we can see no comms for a few seconds, then a bunch of `Not SD printing` within the same second, followed by that message being received again with a periodicity of 1s after the comm blackout. These messages would always be sent before the blackout around the same milisecond timestamp (i.e. around timestamp xx:xx:xx,123, then xx:xx:<xx+1>,125 etc.), followed by the burst of messages after the blackout, followed by again a timestamp which matches the milisecond part of before the interruption (i.e. xx:xx:yy,125). This I would say can rule out that it's not an issue coming from the printer, as the printer is physically transmitting everything with the correct periodicity, just that OctoPrint doesn't get it in time.

Tried these variations on my normal setup but saw no change:
- a RPI Zero 2 W
- a RPI 3A+
- UART on the GPIO header of the PI connected to printer's UART
- USB on PI connected to printer's USB
- enabled `ADVANCED_OK` in Marlin and set `BLOCK_BUFFER_SIZE` to 64 which alleviates the problem by working around it but doesn't fix it entirely
- setting the port as `low_latency` from OctoPrint UI (confirmed it's set correctly by `setserial` command)
- VIRTUAL printer - yes, issue happened even there, see `serial (19).log`. Since I discovered this can be reproduced with the VIRTUAL printer, I only used this for further tests. This can rule out that it's not an issue coming from the printer.
- safe mode 
- change `arm_freq_min=1000` in `/boot/config.txt` to always be at max frequency

Example of this happening can be found in `serial (14).log`, around timestamp 2022-07-24 23:55:44,090
1. Last `Not SD printing` message received was at 23:55:40,040
2. 23:55:40,743 ok received 
3. 23:55:44,090 communication resumes, OctoPrint sends next command
4. 23:55:44,106 to ,118 these are the messages which would presumably have been physically sent between 23:55:40 to 23:55:44

Another example in serial (19).log at 2022-07-25 13:34:50,427.

I have attached a ton of logs either with the actual printer or with the virtual one (can be differentiated by the fact that the virtual one doesn't use `ADVANCED_OK`) and the issue is present in all of them at one or multiple points. I usually search for `timeout`  to find an occurrence or use the regex `ok N[0-9]+ P[1-6][0-9]` to find where the printer buffer is emptying. On the logs done with the VIRTUAL printer (`serial (15).log` onwards) I just scroll through the file to look for gaps of >1s in the timestamps.
All logs are done with the same gcode file but have seen the issue with other gcode files too and never at the same line no.

All of this leads me to believe that there is either some kind of buffering done by the serial port on the kernel/OS side, or something in python's PySerial implementation, or a general stall where the system hangs for a few seconds. The data is exchanged after the few second hangup, showing everything which was sent in the meantime but with a delay. But this hangup is long enough to cause the printer to exhaust its command buffer and stay waiting for new commands, which gives blobs on my prints. I don't know that much about the inner workings of the serial port implementation in PySerial/kernel but I am a programmer so would be happy to help debug.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Ender 3 Pro, SKR mini E3 V3 board with Marlin 2.1 from mainline 

### Browser and version of browser, operating system running browser

doesn't matter, had it happen when starting prints from both PC and phone

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220725135327.zip](https://github.com/OctoPrint/OctoPrint/files/9181291/octoprint-systeminfo-20220725135327.zip)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/9181293/serial.log)
[serial (1).log](https://github.com/OctoPrint/OctoPrint/files/9181294/serial.1.log)
[serial (2).log](https://github.com/OctoPrint/OctoPrint/files/9181295/serial.2.log)
[serial (3).log](https://github.com/OctoPrint/OctoPrint/files/9181296/serial.3.log)
[serial (4).log](https://github.com/OctoPrint/OctoPrint/files/9181297/serial.4.log)
[serial (5).log](https://github.com/OctoPrint/OctoPrint/files/9181298/serial.5.log)
[serial (6).log](https://github.com/OctoPrint/OctoPrint/files/9181299/serial.6.log)
[serial (7).log](https://github.com/OctoPrint/OctoPrint/files/9181300/serial.7.log)
[serial (8).log](https://github.com/OctoPrint/OctoPrint/files/9181301/serial.8.log)
[serial (9).log](https://github.com/OctoPrint/OctoPrint/files/9181302/serial.9.log)
[serial (10).log](https://github.com/OctoPrint/OctoPrint/files/9181303/serial.10.log)
[serial (11).log](https://github.com/OctoPrint/OctoPrint/files/9181304/serial.11.log)
[serial (12).log](https://github.com/OctoPrint/OctoPrint/files/9181305/serial.12.log)
[serial (13).log](https://github.com/OctoPrint/OctoPrint/files/9181306/serial.13.log)
[serial (14).log](https://github.com/OctoPrint/OctoPrint/files/9181307/serial.14.log)
[serial (15).log](https://github.com/OctoPrint/OctoPrint/files/9181308/serial.15.log)
[serial (16).log](https://github.com/OctoPrint/OctoPrint/files/9181309/serial.16.log)
[serial (17).log](https://github.com/OctoPrint/OctoPrint/files/9181310/serial.17.log)
[serial (18).log](https://github.com/OctoPrint/OctoPrint/files/9181311/serial.18.log)
[serial (19).log](https://github.com/OctoPrint/OctoPrint/files/9181312/serial.19.log)

jneilliii, 2022-07-25:
> !!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.

Initial review of the log seems to indicate undervoltage issues. This definitely needs to be addressed.

Florin-Popescu, 2022-07-25:
I did not see any undervoltage issues with the Pi Zero 2 W. Normally my understanding is OctoPrint does show it in the interface if there was any undervoltage since last power on, and never saw it with the Zero 2 W.
I have tried powering the Pi both from the printer motherboard and from an external USB supply. For both I saw no undervoltage.

I did however see undervoltage immediately with the Pi 3 A+ when I tried it with the same external USB supply, did not try it with the printer motherboard supply. So the reporting is definitely there in the interface. What do I make of it if I never saw it with the Zero 2 W?

Would an undervoltage show up in the octoprint.log? I saved a bunch of those too and didn't see anything there, but maybe I'm not looking properly. Not all of them are taken at the same time as the serial.logs unfortunately, but if it would show up there I can start another print right now.

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/9184445/octoprint.log)
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint/files/9184446/octoprint.1.log)
[octoprint (2).log](https://github.com/OctoPrint/OctoPrint/files/9184447/octoprint.2.log)
[octoprint (3).log](https://github.com/OctoPrint/OctoPrint/files/9184448/octoprint.3.log)
[octoprint (4).log](https://github.com/OctoPrint/OctoPrint/files/9184449/octoprint.4.log)
[octoprint (5).log](https://github.com/OctoPrint/OctoPrint/files/9184450/octoprint.5.log)
[octoprint (6).log](https://github.com/OctoPrint/OctoPrint/files/9184451/octoprint.6.log)
[octoprint (7).log](https://github.com/OctoPrint/OctoPrint/files/9184453/octoprint.7.log)
[octoprint (8).log](https://github.com/OctoPrint/OctoPrint/files/9184454/octoprint.8.log)
[octoprint (9).log](https://github.com/OctoPrint/OctoPrint/files/9184455/octoprint.9.log)
[octoprint (10).log](https://github.com/OctoPrint/OctoPrint/files/9184456/octoprint.10.log)
[octoprint (15).log](https://github.com/OctoPrint/OctoPrint/files/9184457/octoprint.15.log)

Florin-Popescu, 2022-07-25:
Just did a test print with the virtual printer and everything else the same. At 21:18:30,617 it happened again in this serial.log.
Attached also octoprint.log where there was a single undervoltage reported but yesterday, and on the Pi3 A+ as stated earlier. No such issue on the Zero 2 W.

Also running `vcgencmd get_throttled` throughout and after that 'print' returns `0x0`, so I think it's highly unlikely to be any undervoltage and also the OctoPrint interface doesn't report anything.

[serial (20).log](https://github.com/OctoPrint/OctoPrint/files/9184553/serial.20.log)

[octoprint (11).log](https://github.com/OctoPrint/OctoPrint/files/9184563/octoprint.11.log)

foosel, 2022-08-01:
Since you say you have a 20min print with which you can reliably reproduce this, I'd love to get my hands on this for further analysis. The fact that it also happens with the virtual printer points to something in OctoPrint or Python to be the culprit here, or maybe some scheduling issues with the OS. PySerial and the serial drivers in the kernel are not involved with the virtual printer. On first look it seems to be 2s pause on the virtual printer and 5 on the real one, does that match your observations? Have you spotted any warnings or such in the syslog for the affected timeslots?

> Would an undervoltage show up in the octoprint.log?

Yes. E.g. here from your first bundle:

```
2022-07-24 21:17:30,089 - octoprint.plugins.pi_support - WARNING - This Raspberry Pi is reporting problems that might lead to bad performance or errors caused by overheating or insufficient power.
!!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.
```

and also couple lines further down in the environment log (look at `plugins.pi_support.throttle_state`):

```
2022-07-24 21:17:30,104 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 382246912
|  os:
|    bits: 32
|    id: linux
|    platform: linux
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model A Plus Rev 1.0
|      octopi_version: 0.18.0
|      octopiuptodate_build: 0.18.0-1.7.3-20220120112925
|      throttle_check_enabled: true
|      throttle_check_functional: true
|      throttle_state: '0x50000'
|  python:
|    pip: 20.3.3
|    version: 3.7.3
|    virtualenv: /home/pi/oprint
```

You can btw also look at your systeminfo bundles using the viewer available at https://bundleviewer.octoprint.org, e.g. [here](https://bundleviewer.octoprint.org/?url=https%3A%2F%2Fgithub.com%2FOctoPrint%2FOctoPrint%2Ffiles%2F9181291%2Foctoprint-systeminfo-20220725135327.zip). Note that currently it also warns that you disabled the throttle check that at least so far I wasn't able to spot in the bundle, so there might still be a bug in that particular detection logic.

closed 2023-01-24 by foosel