OctoPrint/OctoPrint #4611: [Request]

iKrushYou, 2022-08-07:
### Describe the solution you'd like

i was trying to install a plugin for the first time and there is currently a print going

I did not notice the warning saying that plugin management will be disabled while printing right away and spent 5 or so minutes debugging, different browser, searching the internet, reading the HTML source only to finally see the warning and realize that the buttons are intentionally disabled. 

Can the buttons be made more obviously disabled? I suppose it's the theme's doing but the buttons look enabled to me 

Or, there should be a tooltip saying that it's disabled for x reason 

[image]

jneilliii, 2022-08-07:
This may be a moot point in a future release as plugin installs will be able to be queued similar to plugin/OctoPrint updates added in 1.8.0.

cp2004, 2022-08-11:
Addressed with #4620

PR #4620 (2022-09-28): :lipstick: Add tooltip to disabled plugin manager install
Closes #4611

The contrast between enabled/disabled is not great across the board for Bootstrap 2 when there is no enabled button beside it to compare with.
commit bbd06a02 (2022-09-28): :lipstick: Add tooltip to disabled plugin manager install (#4620)

* :lipstick: Add tooltip to disabled plugin manager install

Closes #4611

* :globe_with_meridians: Allow translation of disable string
closed 2023-05-23 by foosel