OctoPrint/OctoPrint #4612: [Request] Different notification mechanism about new plugins on the repository

foosel, 2022-08-08:
### Is your feature request related to a problem? Please describe.

Currently OctoPrint's built in announcement plugin shows popups for new feed items. One of the preconfigured feeds is also the plugin repository's feed, used to notify users about newly registered plugins on the repo right in their OctoPrint instance.

Currently the announcement plugin does this via popups, one popup per feed. This leads to quite a lot of popup spam, especially with regards to the plugin repository.

A recent poll done on OctoPrint's Twitter and Mastodon accounts and on Discord shows that the majority of people *do* like getting notified about new plugins, but the current approach to notification is not well received and considered too spammy. Personally I have to agree.

### Describe the solution you'd like

We need a better approach to provide the information currently provided by the announcement plugin, especially the "new plugin" notifications, that isn't as spammy and annoying as the current popup based approach. Meanwhile important announcements (e.g. security breaches) should still trigger a popup.

One idea that would also combat the general notification spam would be to utilize PNotify's stack functionality to push all notifications into a special notification area instead of the general browser view, and having that only trigger some form of notification icon in the navbar or similar. Important system notices could be redirected to a special "highprio" stack that behaves like before, possibly automatically in case of "error" type notifications. This would be a nuclear approach however as it would also change plugin notification behaviour.

Alternatively we could reimplement the announcement plugin to do the above at least for its news items.

And/or move the plugin repo notices out of the announcement plugin completely and instead into the Plugin Manager as a separate tab or something, turning the whole thing into something the user actually has to seek out. Personally however I think that would be the least favorable approach as it would hide the information from most and likely reduce the likelihood of sudden "oh niiiiice" moments that people quoted as being fun.

### Additional context

Link to polls:

- Twitter: https://twitter.com/OctoPrint3D/status/1554047891467128832
- Mastodon: https://fosstodon.org/@octoprint/108747090579146504
- Discord: https://discordapp.com/channels/704958479194128507/704969133263159296/1003608260739334244

riffnshred, 2022-08-08:
The icon in the navbar would be way better. Simply having a notification icon with a colored dot as an indicator would do the job And if there is a system update have an indicator appear over the settings button.

foosel, 2023-08-29:
Completely forgot to close this, a notification bubble for unread notifications has been implemented as of 1.9.0, anything that isn't from a prio 1 channel (e.g. Important) will no longer trigger a popup.

closed 2023-08-29 by foosel