OctoPrint/OctoPrint #4615: Updating 1.9.0.dev to Flask 2.2 triggers a lot of deprecation warnings regarding `_request_ctx_stack`

foosel, 2022-08-09:
### Problem

Switching `maintenance` to Flask 2.2 logs a lot of warnings [from this line](https://github.com/OctoPrint/OctoPrint/blob/f217ef31963aba0f4a52eedbdea536932ec11ef1/src/octoprint/server/util/flask.py#L95) to the terminal, which is part of the `fixed_get_translations` monkey patch:

```
2022-08-09 14:51:10,079 - py.warnings - WARNING - d:\code\octoprint\octoprint\src\octoprint\server\util\flask.py:95: DeprecationWarning: '_request_ctx_stack' is deprecated and will be removed in Flask 2.3. Use 'g' to store data, or 'request_ctx' to access the current context.
  ctx = _request_ctx_stack.top
```

Additionally, there are some warnings from flask_assets:

```
2022-08-09 14:51:10,478 - py.warnings - WARNING - d:\Code\OctoPrint\devenv37\lib\site-packages\flask_assets.py:273: DeprecationWarning: '_request_ctx_stack' is deprecated and will be removed in Flask 2.3. Use 'g' to store data, or 'request_ctx' to access the current context.
  if not _request_ctx_stack.top:
```

### Solutions

Find a solution to get rid of the warnings. In case of our own code we can modify it as needed. In the case of flask-assets, maybe an update might help, if there is one available.

cp2004, 2022-08-11:
Flask assets seems almost completely unmaintained.

foosel, 2022-08-18:
Indeed it does... For now I've solved it with a monkey patch, long term we might have to consider vendoring or maybe finding an alternative.

I took care of the aforementioned warnings now, and also one I had previously missed about JSONEncoder. Currently everything seems to be fine again on the warning front.

commit 18871db4 (2022-08-18): 🚨 Fix Flask/Werkzeug 2.2 warnings

Closes #4615
closed 2023-05-23 by foosel