OctoPrint/OctoPrint #4617: [Request] restart mjpeg stream automatically

arekm, 2022-08-09:
### Is your feature request related to a problem? Please describe.

Sometime when there is a problem with stream or when webcamd is restarted then stream in octoprint windows stops. Restarting it requires whole UI reload. That can be solved from javascript from what searching tells me.

### Describe the solution you'd like

Polling with image.decode() like described here:

https://stackoverflow.com/questions/67204268/how-to-detect-when-mjpeg-stream-stops-with-javascript

That would allow for stream to recover automatically.

Above url shows such example:

```javascript
<img id="stream" src="http://10.0.0.119:8080/stream"></img>
    <script>
      let image = document.getElementById('stream');

      async function check() {
        while (true) {
          try {
            await image.decode();
          } catch {
            let src = image.src;
            image.src = "";
            image.src = src;
          }
      
          await new Promise((resolve) => setTimeout(resolve, 5000));
        }
      }
      
      check();
    </script>
```

### Additional context

How to reproduce problem - go to "Control" tab, view stream, ssh to pi, sudo ```systemctl restart webcamd```, observe stream stop and no recovery.

foosel, 2022-08-09:
Open to a refresh button here, but an automatism like this without backing off after x failures sounds like a really bad idea to me.

cp2004, 2022-08-09:
Switching tabs in the UI (assuming you've got no plugins that mess with it/webcam stream), with a pause (it waits a second or two) in between should unload the webcam stream, then when you go back it should reload it as well, rather than needing a full UI reload.

arekm, 2022-08-09:
That's very hidden feature! (on first tries years ago I've tried switching, as anyone would, but without waiting. Later I was just doing UI reload automatically, by hand).

So another idea is that image.decode test could be run once, based on tabs switch, instead of automatism or reload button.

cp2004, 2022-08-09:
> That's very hidden feature!

I guess it is for this particular usage - it's designed to unload the stream in the background so it doesn't use a lot of data when not viewing it. It waits some time to avoid problems constantly unloading/reloading  the stream.

I like the idea of a 'refresh' or 'reload' button.

jneilliii, 2022-08-09:
Same thing happens if you minimize your browser window. Minimize the window, wait a second or two and restore it back to view and that should also flip the img src.

foosel, 2023-08-31:
That was an easy win...

[image]

Will be in 1.10.0. Note that it is up to the webcam viewmodel to implement the actual refresh procedure, and for the bundled classic webcam only the mjpg stream type currently supports this. For all other stream types the button is a no-op.

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel