OctoPrint/OctoPrint #4619: A suggestion to adjust decimal precision in some REST responses

nfedyashev, 2022-08-10:
### The problem

I'm using this awesome tool using web and REST HTTP interface.
I'm not sure it's technically a bug.. more like a suggestion to adjust decimal scale, number of digits to the right of the decimal point in a number in HTTP responses.

For example,

```
#+BEGIN_SRC http
GET http://octoprint.local/api/job
Authorization: Bearer ${access-token}
#+END_SRC

#+RESULTS:
#+begin_example
{
  "job": {
    "averagePrintTime": null,
    "estimatedPrintTime": 466.66518850891975,
    "filament": {
      "tool0": {
        "length": 391.50542999999936,
        "volume": 0
      }
    },
    "file": {
      "date": 1660168663,
      "display": "leveling-without-m420.gcode",
      "name": "leveling-without-m420.gcode",
      "origin": "local",
      "path": "leveling-without-m420.gcode",
      "size": 316603
    },
    "lastPrintTime": null,
    "user": "root"
  },
  "progress": {
    "completion": 82.60502901109591,
    "filepos": 261530,
    "printTime": 603,
    "printTimeLeft": 271,
    "printTimeLeftOrigin": "linear"
  },
  "state": "Printing"
}
#+end_example
```

I believe a scale of 2(two digits after to the right of the decimal point) is enough for most(all?) use cases.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.2

### Operating system running OctoPrint

0.18.0

### Browser and version of browser, operating system running browser

HTTP REST

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2022-08-11:
I don't think rounding/truncating the precision to 2 decimal places (or any number of decimal places/significant figures) across the board is correct, because it may be very situation-dependent. The completion progress I believe is used in syncing the gcode viewer, so that needs to be more accurate.

Maybe some of them could be less accurate, but I personally might rather leave it as it is to avoid the chance of issues. I'll leave final decision to Gina on that however.

jneilliii, 2022-08-11:
I agree with Charlie, seems more appropriate for the front end to handle any conversions as it seems fit.

foosel, 2022-08-11:
Yes, throwing away available precision here really is not a good idea.

closed 2022-08-11 by foosel