OctoPrint/OctoPrint #4621: Whitespace not trimmed in username creation

TennisRunner, 2022-08-13:
### The problem

Created an account called "Admin ".
Now every time I log in, I have to include the empty space since it wasn't trimmed during the account creation process. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.1

### Operating system running OctoPrint

Android

### Printer model & used firmware incl. version

Ender 3

### Browser and version of browser, operating system running browser

Firefox or Chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Thats about it

cp2004, 2022-09-06:
As the PR #4631 does - I don't think whitespace should be silently trimmed, but rather not accepted in the first place.

foosel, 2022-09-07:
Agreed, and the input fields for creation in wizard and settings should reflect that accordingly by switching to error state if there's leading or trailing whitespace. Or whitespace in the first place.

040medien, 2022-09-09:
I updated my PR as per the suggested changes. I didn't disallow names with spaces in the middle, because this is not causing issues and seems like a valid use case, e.g. for "Firstname Lastname".

foosel, 2022-10-04:
PR merged (thanks!), will be part of 1.9.0.

commit df2828c8 (2022-10-04): 🥅 Don't accept user names with extra spaces (#4631)

Closes #4621 

Individual commits:

* User names with extra spaces are invalid

* fix return value formatting

* API: Add user name validation

* Core Wizard: Add error message

Add 'invalid user name' error message

* Fix colon typo

* Fix style

* Fix import

* fix error state in core wizard

separately handle invalid user name

* fix isort style error

* Fix API validation

python strings use strip()

* add user name validation to settings

* Correct spelling of username

* implemented foosel's requested changes

changed checks from inval [... cut]
closed 2023-05-23 by foosel