OctoPrint/OctoPrint #4622: [Request] continue failed prints from GCode Viewer

bitranox, 2022-08-16:
### Is your feature request related to a problem? Please describe.

Last Time my print get stucked, because of heatcreep - I could recover printing by manually editing the gcode file and continue printing.
But it would be very nice to have something like a "recovery console" in the GCode Viewer : 

You can Identify (or measure) the height from which on to continue - or just move the extruder slowly to the desired Z-Position.

And maybe some additional buttons for loading Filament, resetting some Axes, resetting the Extruder, slowly move the extruder to the missed layer, etc. to make recovery of a print as conveniant as possible.

### Describe the solution you'd like

a recovery console, for easy recovery of failed prints.

### Describe alternatives you've considered

dio it manually. Its possible, but not for everyone ...

jneilliii, 2022-08-16:
This topic has been covered ad nauseum , here, the forum, and in the Discord server. This is not something that is a trivial task; stepper motors loose position after power loss, the printer's firmware buffer can't be taken into account (gcode viewer is several steps ahead of what is actually being printed by your printer), etc. If anyone wants to tackle this it would be in the form of a plugin and not adapted to core based on previous comments I've seen from foosel.

jneilliii, 2022-08-16:
closing as duplicate: #1876, #1889, #3359

closed 2022-08-16 by jneilliii