OctoPrint/OctoPrint #4626: Babel translation seems not working anymore in python code

NilsRo, 2022-08-20:
### The problem

1. gettext translations does not work in my plugin anymore in python code https://community.octoprint.org/t/babel-does-not-translate-in-python-but-in-javascript-it-is-fine/46134
2. Perhaps caused by: translations generated in python seems not working in general anymore. E.g. the permissions are not translated even that they are included in the message.po of OctoPrint

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.2

### Operating system running OctoPrint

OctoPI and Windows 10 for Development

### Printer model & used firmware incl. version

Anycubic Mega S

### Browser and version of browser, operating system running browser

Edge and Firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220820205809.zip](https://github.com/OctoPrint/OctoPrint/files/9387887/octoprint-systeminfo-20220820205809.zip)

cp2004, 2022-08-20:
I think the issue here is that the translations for `gettext` require a flask context in order to know which language to translate the strings into.

In the case of the permissions - `gettext` is run when the permissions system is started, so they are not translated because there is no flask context at the time. Strings *are* translated from Python (and this is working), but only if they are called as part of serving a request.

Something like `get_template_configs` is called to render the UI, so it has a request context. Any of the `SimpleApiPlugin` stuff, any blueprint stuff, these are called with the request context which means that they know the correct language to translate to. For example:
https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/plugins/action_command_notification/__init__.py#L88-L94

[image]

In your event handler in your plugin, there is no flask context (it's not part of serving any requests or rendering the UI) here to get the target language from either, so it can't be translated.

Unfortunately, I don't see a way around this issue, as multiple users can use different languages, even at the same time - there is logically no way to translate the string if you have one English user and one German user, for example.

I hope that explanation made sense, if not ask questions!

NilsRo, 2022-08-21:
Very good description... 👍  

So the same problem with permission as with my plugin. :-) Will change the issue title later that it is not generic problem but still one with permissions.py.

What would you suggest? My first thoughts was moving the message texts to JS file that should know the context every time...are there better approaches?

cp2004, 2022-08-21:
> What would you suggest? My first thoughts was moving the message texts to JS file that should know the context every time...are there better approaches?

I haven't thought of another solution, other than using an ID to match the translations in the frontend. For OctoPrint, I think this is unlikely to happen but for your plugin, it might work.

NilsRo, 2022-08-22:
> > What

> 
> I haven't thought of another solution, other than using an ID to match the translations in the frontend. For OctoPrint, I think this is unlikely to happen but for your plugin, it might work.

Done it like this and it's fine. The messages are static so it was no problem to move it. Thanks for sharing your thoughts.

foosel, 2023-01-26:
I've added a new `gettext` knockout binding to solve issues like this. As long as the string is wrapped with gettext or _ in the backend, it will get picked up by the translation workflow. Applying `gettext: value` then in the knockout binding will look up the translation for the string and apply it accordingly. No translation means the value will just pipe through as is.

commit 61195f06 (2023-01-26): 🐛 Translate Permission descriptions in UI

Use new gettext binding.

Closes #4626
closed 2023-05-23 by foosel