OctoPrint/OctoPrint #4629: Update to Werkzeug 2.1 causes 400 to be returned on /api/login?passive

crysxd, 2022-08-21:
### The problem

With the latest maintenance branch requests to `/api/login?passive` cause a 400 if the POST request does not have a JSON body. With OctoPrint 1.8 and older, this request works fine:

```
--> POST http://192.168.1.201:8080/api/login?passive http/1.1
 X-Api-Key: 1BF3DBF641EC428CA5946B27EBFAFCE2
 Accept: application/json
 Content-Length: 0
 Host: 192.168.1.201:8080
 Connection: Keep-Alive
 Accept-Encoding: gzip
 User-Agent: okhttp/4.9.2
 --> END POST (0-byte body)
 <-- 400 BAD REQUEST http://192.168.1.201:8080/api/login?passive 
 Content-Type: application/json
 Content-Length: 112
 Cache-Control: no-store, no-cache, must-revalidate, 
 Pragma: no-cache
 Expires: -1
 X-Clacks-Overhead: GNU Terry Pratchett
 Server-Timing: app;dur=8
 X-Robots-Tag: noindex, nofollow, noimageindex
 X-Content-Type-Options: nosniff
 X-Frame-Options: sameorigin
 {
   "error": "Did not attempt to load JSON data because the 
request Content-Type was not 'application/json'."
 }
 <-- END HTTP (112-byte body)
 ```

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

/

### Version of OctoPrint

983685090f9b34c09874d253f0980373f665a4eb

### Operating system running OctoPrint

OctoPi/macOS

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

/

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

See [#dev-core](https://discord.com/channels/704958479194128507/705047010641838211/1010902271183306855)

[Uploading octoprint-systeminfo-20220821144755.zip…]()

cp2004, 2022-08-21:
The entry in the [werkzeug changelog is at the bottom of 2.1.0](https://werkzeug.palletsprojects.com/en/2.2.x/changes/#version-2-1-0), although I think Flask is using 2.2.x by now:

> Request.get_json() will raise a 400 BadRequest error if the Content-Type header is not application/json. This makes a very common source of confusion more visible. [#2339](https://github.com/pallets/werkzeug/issues/2339)

There is a parameter we could add to the `request.get_json()` call in the login endpoint to suppress the 400 and just return None, or handle it differently in OctoPrint (try/except with the `BadRequest` exception)

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/api/__init__.py#L283-L287

https://github.com/pallets/werkzeug/blob/5dff30f8191589ddc5faf1aa2f445fe30befcbcf/src/werkzeug/wrappers/request.py#L546

arekm, 2022-08-21:
Hit the same in firmware update plugin:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/324/commits/b38c2ae64a38db9e61890d1c57b08bc3107455b9

so probably the fix will be
```
data = request.get_json() if request.is_json else request.values
```

cp2004, 2022-08-21:
I searched for other uses of a similar pattern in OctoPrint and found a couple more locations:

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/plugins/corewizard/subwizards.py#L88-L90

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/api/printer_profiles.py#L60-L63

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/api/access.py#L144-L146

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/api/access.py#L198-L201

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/api/access.py#L249-L252

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/api/access.py#L299-L302

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/api/printer.py#L402-L405

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/server/util/flask.py#L1546-L1550

All but the top one are likely non-issues as the code would just never get to manually returning a 400.

I have run a scan against the current versions of all plugins in the plugin repository to find usages of request.json or request.get_json and am confident that there was only one other plugin that may be caught out by this, and I have opened an issue.

crysxd, 2022-08-22:
Maybe even a good idea to remove the manual `is None` check as it will never trigger again and rely on Wekrzeug's check?

foosel, 2022-08-22:
For the record, leaning towards that as well here, it's cleaner.

Btw, `/api/login?passive` working definitely was a bug rather than a feature, the [API documentation](https://docs.octoprint.org/en/master/api/general.html#login) states that `passive` is a JSON parameter, and that [unless otherwise noted all requests should use an `application/json` body](https://docs.octoprint.org/en/master/api/general.html#content-type).

foosel, 2022-09-28:
Fixed via #4630 (thanks @cp2004!)

closed 2022-09-28 by foosel