OctoPrint/OctoPrint #4632: [Request] - SimpleApiPlugin Mixin Caller Permissions

QuinnDamerell, 2022-08-26:
### Is your feature request related to a problem? Please describe.

The SimpleApiPlugin mixin is a handy way to make simple API calls easily; I love it. But something I noted that would be a nice improvement would be the ability to control users' permissions on a per-command or GET call basis.

Right now, the only possible level of permission checking is using the `is_api_adminonly`, which only allows very course control; it's all or nothing. Beyond that, there's no way to check the caller's permissions since no caller information is passed beyond the command and data.

The biggest issue I have run into is that I would like to protect my APIs so users with access to OctoPrint can access them, but callers with no user context can't. I know the user system in OctoPrint isn't considered fully secured, but it would at least add some level of security.

### Describe the solution you'd like

A nice addition to the API would be to add more permission-checking capabilities. I was thinking of two ways of doing it:

1. In the get_api_commands dict returned, some kind of permission level could be passed for each command.
2. Additional caller permission levels could be passed to the `on_api_command` and `on_api_get,` so the logic handling the request can determine if the permissions are valid.

The issue with #1 is it doesn't provide any way of setting permissions for the GET call. But it's simpler than #2, so I still like it.

cp2004, 2022-08-26:
You can check the permissions easily yourself using `octoprint.access.Permissions` - for example:

https://github.com/OctoPrint/OctoPrint/blob/ae1cba8d914f390dfa13ea569e23a2f1d30d3384/src/octoprint/plugins/pluginmanager/__init__.py#L638-L650

I am not sure how well they are documented specifically for plugin authors - the documentation rewrite in-progress will have a better guide on how to use the permission system (or at least I'm planning to write it!).

QuinnDamerell, 2022-08-26:
@cp2004  oh that's great as well! That might do the trick for what devs need. I don't think I saw that anywhere in the docs, it would great to add in there!

jneilliii, 2022-08-26:
Yeah, I use this approach in a couple of my plugins. Here's another example with custom permissions. 

https://github.com/jneilliii/OctoPrint-OctoDashCompanion/blob/faae265de9c7b83b5e2754edad4be649b9c40f26/octoprint_octodashcompanion/__init__.py#L229-L231

cp2004, 2022-08-26:
I opened https://github.com/OctoPrint/docs/issues/4 to track the need to write a guide here for the docs-rewrite.

closed 2022-08-26 by cp2004

QuinnDamerell, 2022-08-26:
Thanks @cp2004 !